platformio/platformio-vscode-ide #1852: uint8_t and uint32_t get mapped wrong by Intellisense for Arduino Nano 33 BLE

Alex-Evo, 2020-07-06:
If I create a new project for an "Arduino Nano 33 BLE" and create test variables of the "stdint" types, I get the following situation:

[image]

Hovering over the types gives the following results:
**int8_t**: typedef __int8_t int8_t => OK

**uint8_t**:
```
<error-type> uint8_t
@brief Function for reading data from the SPI receiver register.
@param[in] p_reg Pointer to the peripheral registers structure.
@return RX data received.
variable "uint8_t" is not a type name
```

F12 takes me to line 228 "[...]/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/modules/nrfx/hal/nrf_spi.h"

This does not make any sense, but it looks like Intellisense maps the definition of uint8_t to the return type of the macro defined in that line.

**int16_t, uint16_t, int32_t**: all OK

**uint32_t**:
```
typedef <error-type> uint32_t(<error-type>)
! The RND Generate vector function pointer type definition.
The prototype intendent for External and CRYS internal RND functions
pointers definitions.
Full description can be found in ::CRYS_RND_GenerateVector function API.
```

F12 takes me to line 206 of "[...]/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino/mbed/features/cryptocell/FEATURE_CRYPTOCELL310/include/crys_rnd.h"

Also here, it looks like Intellisense maps the definition of uint32_t to the return type of the macro defined in that line.

Any help (can also be hacks or workarounds) would be highly appreciated. From past issues, it looks like @sean-mcmanus has been knowledgeable about these topics in the past...

I'm running the following configuration:
PlatformIO Core: 4.3.4
PlatformIO Home: 3.2.3
ms-vscode.cpptools: 0.28.3 (also tried downgrading to 0.26.0 with same result)
VS Code version info:
Version: 1.46.1
Commit: cd9ea6488829f560dc949a8b2fb789f3cdc05f5d
Date: 2020-06-17T21:17:14.222Z (2 wks ago)
Electron: 7.3.1
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Darwin x64 19.5.0

sean-mcmanus, 2020-07-06:
Are you able to locate the source where uint8_t and uint32_t are defined? It's possible there's some incorrect define or intelliSenseMode being used that causes the definition to be skipped in the headers. Info from running C/C++: Log Diagnostics (with the source file active) might help. If you could provide more detailed repro steps I could try the repro. You may also want to try 0.29.0-insiders (https://github.com/microsoft/vscode-cpptools/releases/tag/0.29.0-insiders ).

Alex-Evo, 2020-07-06:
Thanks @sean-mcmanus for the super-fast reply!

With 0.29.0-insiders, code formatting on save appears to be broken (my source file takes forever to save, and there is message popping up that allows me to cancel formatting - only then the file is actually saved), so I reverted back to the "non-insiders" version.

The defines that work correctly are from "[...]/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/sys/_stdint.h" - I assume that's also where the ones for uint8_t and uint32_t should come from, and I don't know why they are skipped and instead take over the wrong definitions from the two other files I mentioned above.

I changed logging from error to debug, and this is what I got for the source file:
<details><summary>Details</summary>
<p>

```
File exclude: **/.git
File exclude: **/.svn
File exclude: **/.hg
File exclude: **/CVS
File exclude: **/.DS_Store
File exclude: **/.vscode
Search exclude: **/node_modules
Search exclude: **/bower_components
Search exclude: **/*.code-search
Search exclude: **/.vscode
IntelliSense Engine = Default.
Enhanced Colorization is enabled.
Error squiggles are enabled if all header dependencies are resolved.
Autocomplete is enabled.
File exclude: **/.git
File exclude: **/.svn
File exclude: **/.hg
File exclude: **/CVS
File exclude: **/.DS_Store
File exclude: **/.vscode
Search exclude: **/node_modules
Search exclude: **/bower_components
Search exclude: **/*.code-search
Search exclude: **/.vscode
Populate include completion cache.
Discovering files...
  Processing folder (recursive): /Users/alex/.platformio/packages/toolchain-gccarmnoneeabi/lib/gcc/arm-none-eabi/8.2.1/include/
  Processing folder (recursive): /Users/alex/.platformio/packages/toolchain-gccarmnoneeabi/lib/gcc/arm-none-eabi/8.2.1/include-fixed/
  Processing folder (recursive): /Users/alex/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/
  Processing folder (recursive): /Users/alex/Documents/PlatformIO/Projects/200704-161628-nano33ble/include/
  Processing folder (recursive): /Users/alex/Documents/PlatformIO/Projects/200704-161628-nano33ble/src/
  Processing folder (recursive): /Users/alex/Documents/PlatformIO/Projects/200704-161628-nano33ble/.pio/libdeps/nano33ble/Arduino_LSM9DS1_ID6589/src/
  Processing folder (recursive): /Users/alex/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/Wire/
  Processing folder (recursive): /Users/alex/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/PDM/src/
  Processing folder (recursive): /Users/alex/Documents/PlatformIO/Projects/200704-161628-nano33ble/.pio/libdeps/nano33ble/SdFat_ID322/src/
  Processing folder (recursive): /Users/alex/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/SPI/
  Processing folder (recursive): /Users/alex/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino/
  Processing folder (recursive): /Users/alex/.platformio/packages/framework-arduino-nrf52-mbedos/variants/ARDUINO_NANO33BLE/
  Processing folder (recursive): /Users/alex/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/USBHID/
  Processing folder (recursive): /Users/alex/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/USBMSD/
  Processing folder (recursive): /Users/alex/.platformio/packages/tool-unity/
  Processing folder (recursive): /Users/alex/.platformio/packages/toolchain-gccarmnoneeabi/lib/gcc/arm-none-eabi/8.2.1/include/
  Processing folder (recursive): /Users/alex/.platformio/packages/toolchain-gccarmnoneeabi/lib/gcc/arm-none-eabi/8.2.1/include-fixed/
  Processing folder (recursive): /Users/alex/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/
  Processing folder (recursive): /Users/alex/Documents/PlatformIO/Projects/test/include/
  Processing folder (recursive): /Users/alex/Documents/PlatformIO/Projects/test/src/
  Processing folder (recursive): /Users/alex/.platformio/packages/framework-arduino-nrf52-mbedos/cores/arduino/
  Processing folder (recursive): /Users/alex/.platformio/packages/framework-arduino-nrf52-mbedos/variants/ARDUINO_NANO33BLE/
  Processing folder (recursive): /Users/alex/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/PDM/src/
  Processing folder (recursive): /Users/alex/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/SPI/
  Processing folder (recursive): /Users/alex/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/USBHID/
  Processing folder (recursive): /Users/alex/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/USBMSD/
  Processing folder (recursive): /Users/alex/.platformio/packages/framework-arduino-nrf52-mbedos/libraries/Wire/
  Processing folder (recursive): /Users/alex/.platformio/packages/tool-unity/
  Discovering files: 8621 file(s) processed
  0 file(s) removed from database
Done discovering files.
Parsing remaining files...
  Parsing: 0 files(s) processed
[... 919 lines cut ...]
  define: TOOLCHAIN_GCC_ARM
  define: WSF_MAX_HANDLERS=10
  define: MBED_NO_GLOBAL_USING_DIRECTIVE=1
  define: USE_ARDUINO_PINOUT
  define: ARDUINO=10810
  define: ARDUINO_ARCH_MBED
  other: --g++
  other: --gnu_version=80201
Checking for syntax errors: file:///Users/alex/Documents/PlatformIO/Projects/test/src/main.cpp
Queueing IntelliSense update for files in translation unit of: /Users/alex/Documents/PlatformIO/Projects/test/src/main.cpp
idle loop: reparsing the active document
Checking for syntax errors: file:///Users/alex/Documents/PlatformIO/Projects/test/src/main.cpp
[... cut ...] of: /Users/alex/Documents/PlatformIO/Projects/test/src/main.cpp
Error squiggle count: 1
terminating child process: 3447
Update IntelliSense time (sec): 13.297
idle loop: reparsing the active document
Checking for syntax errors: file:///Users/alex/Documents/PlatformIO/Projects/test/src/main.cpp
Queueing IntelliSense update for files in translation unit of: /Users/alex/Documents/PlatformIO/Projects/test/src/main.cpp
Error squiggle count: 1
Update IntelliSense time (sec): 0.257
```

</p>
</details>

Alex-Evo, 2020-07-06:
In case you want to try to reproduce, here is what I did (on macOS 10.15.5):
- Install current version of Arduino IDE (1.8.13)
- Install VS Code, PlatformIO, ms-vscode.cpptools (versions as listed in first post)
- In PlatformIO home: select "New Project" -> Choose "Arduino Nano 33 BLE" as board, keep "Use default location" on
- In main.cpp paste the following code between "#include <Arduino.h>" and "void setup()":

```
int8_t test1;
uint8_t test2;
int16_t test3;
uint16_t test4;
int32_t test5;
uint32_t test6;
```

- Observe the behavior as described in my initial post

sean-mcmanus, 2020-07-06:
I hit the problem with 0.29.0-insiders: https://github.com/microsoft/vscode-cpptools/issues/5731 . Thanks for reporting.

I also hit the issue with uint8_t on 0.28.3 -- changing the intelliSenseMode to clang-arm fixed the issue for me -- can you try that? I think it's a bug with PlatformIO -- they generate the c_cpp_properties.json and they should be setting the intelliSense mode to arm types for arm platforms.

sean-mcmanus, 2020-07-06:
Actually, using clang-arm doesn't fix it (it could fix other issues though), but adding "__STATIC_INLINE" to the defines list appears to have fixed it. Not sure if the PlatformIO extension is supposed to add that automatically or if something else is causing that to not get defined.

Alex-Evo, 2020-07-07:
@sean-mcmanus Great, thanks so much! I can confirm that adding adding "__STATIC_INLINE" to the defines list in "c_cpp_properties.json" fixes the issue also for me!

If I understand the situation correctly, this means that PlatformIO needs to change its behavor in two ways:
- Set "intelliSenseMode" to "clang-arm" for ARM platform projects
- Make sure that "__STATIC_INLINE" is included in the defines list

I hope they can adapt that in one of the next releases...

Alex-Evo, 2020-07-07:
Just a note for other noobies like me: As PlatformIO regenerates the "c_cpp_properties.json" file, you need to tell PlatformIO to include the respective define for you using the "build_flags" option, i.e. a minimal working "platformio.ini" for an Arduino Nano 33 BLE looks like this:

```
[env:nano33ble]
platform = nordicnrf52
board = nano33ble
framework = arduino
build_flags =
    -D__STATIC_INLINE
```

sean-mcmanus, 2020-07-07:
Hmm, the clang-arm setting causing missing headers is not expected. Maybe it should be clang-arm64? If the target is 32-bit, it'd be interesting to see if clang-x86 works better or not.

sean-mcmanus, 2020-07-09:
FYI, https://github.com/microsoft/vscode-cpptools/releases/tag/0.29.0-insiders2 has the fix for the bug you hit with 0.29.0-insiders, as well as some other fixes we found while working with PlatformIO-based projects.

Alex-Evo, 2020-07-09:
Thanks @sean-mcmanus that's great news. I'll give it a try (and also provide feedback on the clang topic) as soon as I can continue to work on our project (I'm currently held up by other priorities, but should be able to return soon)

Alex-Evo, 2020-07-13:
@sean-mcmanus: Thanks again. I can confirm that 0.29.0-insiders2 fixes the bug for me as well.

clang-x86 (instead of clang-arm, which didn't) seems to work, however, PlatformIO overwrites the setting back to clang-x64 when it is regenerating c_cpp_properties.json - and as I had no notable issues with clang-x64 in the first place, I am not particularly worried about that...

Alex-Evo, 2020-07-13:
Oh, wait, this is new... Now I'm getting an "include error detected" message, when I use clang-x64. It goes away with clang-x86, but I don't know how to make this setting persistent, as noted in my previous post 🙈

Alex-Evo, 2020-07-13:
... and more annoying, I get lots of build errors if I add

```
build_flags =
    -D__STATIC_INLINE
```

to my platformio.ini

Alex-Evo, 2020-07-13:
So, the only thing that currently works is manually adapting  c_cpp_properties.json to add
"__STATIC_INLINE" to the "defines" list and to change
"intelliSenseMode" to "clang-x86".

However, I will lose both modifications whenever PlatformIO chooses to recreate c_cpp_properties.json

sean-mcmanus, 2020-07-13:
The C/C++ extension added clang-x86 mode for 32-bit targets in August 2019 -- not sure if PlatformIO has added support for that yet.

I also hit a problem using clang-arm mode -- looks like we force a --target=arm-arm-none-eabi for 32-bit ARM compiler querying as a default, which causes some compilers that don't support that target to exclude all the platform includes -- adding a correct `--target=<triplet>` to the compilerArgs should fix that. I've filed an issue at https://github.com/microsoft/vscode-cpptools/issues/5772 .

sean-mcmanus, 2020-08-06:
https://github.com/microsoft/vscode-cpptools/releases/tag/0.30.0-insiders added a bunch of updates to auto-handling of ARM intelliSenseModes and added remote ARM support, so let us know if the configuration has gotten worse or better for you. In particular, your workaround of using clang-x86 for ARM may no longer work if we detect the compiler is targetting ARM and for it to clang-arm.

sean-mcmanus, 2020-09-14:
I don't think the issue is "stale"...it was just never addressed. What details were needed?