platformio/platformio-vscode-ide #1883: Arduino.h: No such file or directory

doa232, 2020-07-18:
I want to compile a program using Platform IO in VS Code, but keep getting this error
[image]
[image]

I tried everything from the deleting the .vscode and tried bunch of things , please help

54696d21, 2020-07-18:
for what board are you trying to build? Are you even using Arduino as framework? What project are you trying to build (link to it if it is opensource)? Post your platformio.ini file please

doa232, 2020-07-18:
Hi sorry for the late response, 

> for what board are you trying to build? A

Teensy

>Are you even using Arduino as framework? 

I think so

>What project are you trying to build (link to it if it is opensource)? 

It's a school project

>Post your platformio.ini file please

`[env:teensy31]`
`platform = teensy`
`board = teensy31`
`framework = arduino`

Someone suggest that I did not include the library correct, but I don't get it ......

54696d21, 2020-07-18:
1. please share your code so I can have a look (if possible)
2. Try to create a new project in PlatformIO and compile it empty and see if the problem persists
3. try to reinstall PlatformIO
4. if you can't share your code and 2 and 3 didn't help, ask your supervisor in school

We can continue in german btw if that's easier for you (either is fine with me)

doa232, 2020-07-18:
Hi, I found the problem turns out I was clicking Run the whole time instead of Build. Now it compiles as it should be, sorry for being a moron,
But I really thought that clicking the white triangle button for Run will build/compile the code first, isn't that how it usually works? Also, does that mean it can be compiled but there's  still could be errors? 
[image]

I am not German btw but I am studying in Germany, how did you know? ( It's easier for me in English , my German is average at best)

54696d21, 2020-07-18:
I saw the comments in the source code are German. Your code can of course still have errors, but no more syntax error that the compiler can catch. I use the shortcut Ctrl+shift+B to compile and Ctrl+shift+U to load it to the board.

doa232, 2020-07-18:
Understood,  I will go back here again if encounter more problem
thanks again for your time

closed 2020-07-18 by doa232