platformio/platformio-vscode-ide #1886: Platformio extension missing from Vscodium and code-oss

LaBus3, 2020-07-18:
Last checked on 4:39:24 PM July 18, 2020 GMT. Available and working on mainstream vscode builds.

Pop!_OS 20.04 LTS

closed 2020-09-21 by stale[bot]

nicoduck, 2020-10-03:
This is still the case. Installaing PlatformIO manually from brings up more errors, some cpp dependencies cannot be found.

fri-sch, 2020-10-10:
It seems like platformio-vscode-ide depends on some extension called `ms-vscode.cpptool` which is not available in the OSS version, but only in the proprietary Microsoft release. This is very unfortunate and should be resolved so Code OSS users can install PlatformIO.

fri-sch, 2020-10-10:
For completeness, this is the message shown when trying to install the extension on Code OSS manually:
```
Cannot activate the 'PlatformIO IDE' extension because it depends on an unknown 'ms-vscode.cpptools' extension .
```

fri-sch, 2020-11-01:
Can someone please remove the `stale` label and reopen this issue, thanks!

As for a workaround that seems to work for me:
* Install the extension from file
* Open `~/.vscode-oss/extensions/platformio.platformio-ide-2.1.0/package.json`
*

clwillingham, 2020-11-12:
can confirm, removing `ms-vscode.cpptools` from `extensionDependencies` works. would this break anything? looking at git blame suggests it was added for the PIO Unified debugger. could anyone who uses this feature test whether that feature still needs that file?

miguelpduarte, 2020-11-30:
I am also facing this issue right now. When searching or using the `ext install` command directly, the extension is not found.

appanp, 2021-01-28:
My laptop running Manjaro linux (type of ArchLinux) is also not able to find the PlatformIO extension from VSCode ,ver 1.51.1 - has to do with some VSCode config setting of where in the internet it is looking for these extensions, not sure which JSON file it is in ?? VSCode in another Ubuntu laptop is able to find the extension from VSCode itself !!

So, I installed it after downloading the extension VIX file from the VSCode marketplace and this gave the error: depends on an unknown 'ms-vscode.cpptools' extension". But I found the extension in the microsoft vscode-cpptools repo (https://github.com/microsoft/vscode-cpptools/releases/tag) from where I downloaded & installed this dependent VIX file.

After this PlatformIO successfully installed and I am able to use the extension. Looks like a pretty rounded extension - not sure if it is good-enough (or may be better than) Arduino IDE to be my only embedded dev platform !!

MRCWirtz, 2021-05-24:
Hi @appanp, for me the issue ocured for Manjaro linux as well, thanks for your hint!
For me the `cpptools-linux.vsix` from tag 1.3.1 worked fine:
https://github.com/microsoft/vscode-cpptools/releases/tag/1.3.1

sschueller, 2022-01-02:
Here are the exact steps to get it to work in Manjaro (Arch):

1. Download and install vscodium 
2. Download cpptools-linux.vsix : https://github.com/microsoft/vscode-cpptools/releases/download/1.3.1/cpptools-linux.vsix
3. Install cpptools-linux.vsix in vscodium under extensions / install from vsix
4. Download PlatformIO vsix: https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide (download button on right side) 
5. Install PlatformIO vsix in vscodium under extensions / install from vsix
6. Add udev rules ```curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core/master/scripts/99-platformio-udev.rules | sudo tee /etc/udev/rules.d/99-platformio-udev.rules```
7. Add your user to the device group: ```sudo useradd myusername uucp```
8. Logout / Login

SarenT, 2022-12-31:
Why is this closed? The issue persists.

duck7000, 2023-01-26:
@sschueller 
the download path gives 404, the realpath/newpath:
curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core/develop/platformio/assets/system/99-platformio-udev.rules | sudo tee /etc/udev/rules.d/99-platformio-udev.rules

All information about udev rules can be found here:
https://docs.platformio.org/en/stable//core/installation/udev-rules.html

sschueller, 2023-01-26:
Th

> @sschueller The download path gives 404, the realpath/newpath:
>
> `curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core/develop/platformio/assets/system/99-platformio-udev.rules | [...]

Thanks, I have updated my comment.

wrickert, 2023-06-17:
Problem still exists

hannescam, 2023-08-05:
i have the same problem

alextrical, 2023-10-16:
Likewise, still an issue for VSCodium

atomspring, 2023-10-29:
Same, let's get thsi working for people who don't owe microsoft s#!t

duck7000, 2023-10-29:
There is a solution so calling "i have the same problem" or "the problem still exists" is not doing  anything.

Some cpp library's can't be included because of rights so if you use code-oss you have to accept that and install them manually if you want to use them, there is no other way around this.

Why Microsoft those libs not open source releases is another question though

calhoward, 2023-12-05:
I can confirm that @sschueller's solution works on Arch as of Dec '23

evren320, 2024-01-19:
In fact Microsoft cpptools are open source and MIT licensed, only the binaries compiled for Microsoft marketplace are proprietary, so it cannot be directly linked by projects other than their own vscode, but anyone can compile cpptools and install it to satisfy depencency requisite for platformio, it works.

MarkJeronimus, 2024-02-09:
Someone with experience in packaging extensions (not me) should compile and publish it, so platformio can get unbroken.

duck7000, 2024-02-09:
Even if someone compile this extension you can not combine it within platformIO, only if you compile it yourself you can use it.

So 2 legal options remain:
1e download the binary yourself as described by others above.
2e compile the extension yourself on your system like @evren320 correctly mentioned.

sschueller, 2024-02-09:
There is a 3rd option. You could turn what I did above into a script and make the process a bit simpler.

XJDHDR, 2024-02-15:
The reason PLatformIO is not installable in VSCodium is because it hasn't been uploaded to the OpenVSX gallery. It can't be uploaded to OpenVSX because a dependancy (`cpptools`) also hasn't been uploaded to OpenVSX.

So as far as I can see it, the hands of the PlatformIO devs are tied right now, and to move forward requires one of the following to happen:

- Modify PlatformIO to get rid of the `cpptools` dependancy, and replace it with something equivalent that is available on OpenVSX, or...
- Upload `cpptools` to OpenVSX, and keep it updated.

kRHYME7, 2024-04-14:
https://github.com/platformio/platformio-vscode-ide/issues/1886#issuecomment-1405598591

Will this work for code-oss too ? 

Thank you

duck7000, 2024-04-14:
@kRHYME7 yes it will

kRHYME7, 2024-04-14:
I only did (archlinux) ``` yay -Sy code-marketplace ``` and I can now access the extensions. https://aur.archlinux.org/packages/code-marketplace

Any advice for the best practices?

clytle374, 2024-04-27:
Been fighting this today myself, sadly it's hard to get to this answer.  Bait and switch, knew we were screwed when MS got into opensource.  Time to move on

KyleMaas, 2024-05-21:
@sschueller Thank you!  Your directions worked on Gentoo.

vinibali, 2025-01-11:
Here is the latest URL for the latest vsix file:
```
https://marketplace.visualstudio.com/_apis/public/gallery/publishers/platformio/vsextensions/platformio-ide/3.3.3/vspackage?targetPlatform=win32-x64
```
Note: it didn't show me with Firefox default privacy settings.

Vzor-, 2025-01-28:
Unfortunately @sschueller's solution no longer works on Arch as of Jan 2025. There is no longer a link to download vsix files from the marketplace.

vinibali, 2025-01-28:
> Unfortunately [@sschueller](https://github.com/sschueller)'s solution no longer works on Arch as of Jan 2025. There is no longer a link to download vsix files from the marketplace.
> 
> I had to use https://marketplace.visualstudio.com/_apis/public/gallery/publishers/platformio/vsextensions/platformio-ide/3.3.4/vspackage?targetPlatform=linux-x64

The version 3.3.3- on my previous comment - was installed on Arch.

jedahan, 2025-05-04:
for macos https://marketplace.visualstudio.com/_apis/public/gallery/publishers/platformio/vsextensions/platformio-ide/3.3.4/vspackage?targetPlatform=darwin-arm64

sschueller, 2025-05-04:
> Unfortunately [@sschueller](https://github.com/sschueller)'s solution no longer works on Arch as of Jan 2025. There is no longer a link to download vsix files from the marketplace.
>
> I had to use [...]

Thank you for the update. I updated my comment above.