platformio/platformio-vscode-ide #1896: Cannot install PlatformIO for VSCode - Start PIO Home Server

SaxonRah, 2020-07-22:
%23 Description of problem
  I've looked all over and tried everything related. I even uninstalled all of my related and unrelated tool-chains, compilers, cleaned path, and deleted everything to do with programming on this machine yet still no dice.

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.47.2
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Error: Could not find a version that satisfies the requirement '~2.39.0' for your system 'windows_amd64'

	at c:\Users\LordRah\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:11965
	at ChildProcess.d (c:\Users\LordRah\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:8215)
	at ChildProcess.emit (events.js:203:13)
	at ChildProcess.cp.emit (c:\Users\LordRah\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

SaxonRah, 2020-07-22:
After having to uninstall literally everything, over 160gb of programs, cleaning the registry, and rebooting several times. I was finally able to see PIO Home and install PlatformIO for VSCode. 

Not really a solution but.... whatever.

IsaacBuilder28, 2020-07-23:
This is how I have been doing it for the past few months, can you guys create a solution to this, the platform idea seems strong and meaningful, but the platform needs much more ironing, and fixing, good luck!

closed 2020-09-21 by stale[bot]