platformio/platformio-vscode-ide #1923: Impossible to install

trevor-sonic, 2020-08-06:
Issue Type: <b>Bug</b>

I'm installing the PlatformIO extension.
Everytime it is saying "PlatformIO IDE has been succesfully installed... [RELOAD]"
Then after reload (can't install bay the way)  it is retrying to install and so on... infinite..

I update Python to 2.7 nothing has changed.

Extension version: 1.10.0
VS Code version: Code 1.47.3 (91899dcef7b8110878ea59626991a18c8a6a1b3e, 2020-07-23T13:08:29.692Z)
OS version: Darwin x64 19.5.0

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-4278U CPU @ 2.60GHz (4 x 2600)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>metal: disabled_off<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_renderer: disabled_off_ok<br>video_decode: enabled<br>viz_display_compositor: enabled_on<br>viz_hit_test_surface_layer: disabled_off_ok<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|2, 9, 6|
|Memory (System)|8.00GB (1.76GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details>

lsaranto, 2020-08-13:
Same issue here.

trevor-sonic, 2020-09-19:
PlatformIO 2.7.2 could finally be installed and it is working.

flashbackpa, 2020-09-28:
> PlatformIO 2.7.2 could finally be installed and it is working.

Hi, sorry where did you find ver 2.7.2 ? On the market the version I found (and that won't work cause of python errors) is PlatformIO IDE 2.1.0.

Thank you

trevor-sonic, 2020-10-04:
> Hi, sorry where did you find ver 2.7.2 ? On the market the version I found (and that won't work cause of python errors) is PlatformIO IDE 2.1.0.

I'm sorry v2.1.0 is what I have installed. You're right.

closed 2020-10-04 by trevor-sonic