platformio/platformio-vscode-ide #1940: [feature request] use clangd as language server

noah1510, 2020-08-16:
At the moment [ms-vscode.cpptools](https://marketplace.visualstudio.com/items?itemName=ms-vscode.cpptools) is used to provide the language server, however this is not fully open source (see [this issue](https://github.com/Microsoft/vscode-cpptools/issues/21#issuecomment-248349017)).

To have this extention fully open source, switching to [llvm-vs-code-extensions.vscode-clangd](https://open-vsx.org/extension/llvm-vs-code-extensions/vscode-clangd) would be great, since that extention is open source on every platform and should be compatible since platformio should be capable of gernerating a compile-commands.json file.

PS. The copyright notice is missing in the license (it still has the placeholder).

closed 2020-09-22 by stale[bot]

yunuscukran, 2020-10-13:
Hi Is there a way to reopen this issue? I'd argue it's a very important feature that would enable PlatformIO to work on Theia, VSCodium, and any other VSC based open-source IDE's...

Trass3r, 2021-08-31:
> platformio should be capable of gernerating a compile-commands.json file

Unfortunately the [clang-tidy integration](https://docs.platformio.org/en/latest/plus/check-tools/clang-tidy.html) was done by manually calling the tool instead of generating a `compile_commands.json` file (which would also allow on-the-fly analysis).
[Generating one manually](https://scons.org/doc/production/HTML/scons-user/ch27.html) is possible via `pio run -t compile_commands.json` with:

`compiledb.py`:
```py
Import("env")
env.Tool('compilation_db')
env.CompilationDatabase('compile_commands.json')
```

```ini
extra_scripts =
  pre:compiledb.py
```

eliliam, 2023-06-02:
Bumping this as a huge chunk of developers are exclusively using the clangd extension over the default C/C++ extension. Please support both!

jmjoy, 2025-03-27:
I want to eliminate the dependency on the C/C++ extension because I only use clangd, and I’d rather manually generate the compile_commands.json file.

Trass3r, 2025-03-27:
See https://github.com/platformio/platformio-vscode-ide/issues/2674#issuecomment-2009452307

jmjoy, 2025-03-27:
But currently it's mandatory to depend on C/C++ extensions, which I really don't want to use at all.

psolyca, 2025-04-06:
Thanks for this tip.
Currently, VSCodium can not use anymore the C/C++ extension. Microsoft blocks other IDE.
https://github.com/VSCodium/vscodium/issues/2300
Could be the rigth time to change the language server.
BTW, there is the debugging to handle.

For informaiton : https://www.reddit.com/r/vsCodium/comments/w9kaag/what_is_the_equivalent_of_the_ms_c_extension_with/