platformio/platformio-vscode-ide #1948: Monitor stays open if debugger is started, despite autoCloseSerialMonitor setting

Panometric, 2020-08-21:
Even if "platformio-ide.autoCloseSerialMonitor": true, running the program from the debugger will not close the monitor. 

This is especially problematic if the board is using JTAG debug probe and also uploading via serial, because it makes the serial fail with:
```
File "c:\users\mike\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM4': PermissionError(13, 'Access is denied.', None, 5)
*** [upload] Error 1
```

See: https://community.platformio.org/t/upload-by-serial-and-debug-using-jtag-dont-sharer-port-well/15450

Panometric, 2020-09-21:
Not stale, still a pertinent issue.

Panometric, 2020-11-28:
Still present in Core 5.0.4a2·Home 3.3.1

```
Linking .pio\build\esp32_wrover_16M_FT2232H\firmware.elf
Retrieving maximum program size .pio\build\esp32_wrover_16M_FT2232H\firmware.elf
Building .pio\build\esp32_wrover_16M_FT2232H\firmware.bin
Checking size .pio\build\esp32_wrover_16M_FT2232H\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   7.9% (used 42080 bytes from 532480 bytes)
Flash: [==        ]  16.3% (used 1070112 bytes from 6553600 bytes)
esptool.py v2.6
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: COM13
Uploading .pio\build\esp32_wrover_16M_FT2232H\firmware.bin
esptool.py v2.6
Serial port COM13
Traceback (most recent call last):
  File "C:\Users\Mike\.platformio\packages\tool-esptoolpy\esptool.py", line 2959, in <module>
    _main()

```

digaus, 2021-05-08:
> Still present in Core 5.0.4a2·Home 3.3.1
>
> ```
> Linking .pio\build\esp32_wrover_16M_FT2232H\firmware.elf
> Retrieving maximum program size .pio\build\esp32_wrover_16M_FT2232H\firmware.elf
> Building [...]

I am also getting the error. Is there a way to fix this?

pearson, 2022-09-27:
This is a problem I've seen whenever using PlatformIO and flashing via serial. 

If I've used the serial monitor, performing an upload will usually generate a `serial.serialutil.SerialException 'COMx': PermissionError(13, 'Access is denied.', None, 5)` on the first try. Trying an upload a second time usually works, although it occasionally requires a third try. 

On rare occasions it will work as expected, and automatically close the serial monitor. It seems like this will happen more often when I've rebooted the PC and started working with Visual Studio Code, but I haven't proven that to myself. (I'll start working with PlatformIO for the first time in a while, and uploads/using the serial monitor seem to work correctly, but I'll eventually start seeing this error and think "oh, right...")

I was quite surprised to see it being an issue when flashing via JTAG, with a separate serial connection for the serial monitor. This doesn't seem to make much sense to me, but it clearly happens. For this configuration I'm using an ESP32-PROG for JTAG, and I've tried both an FT232 and the ESP32-PROG's serial port for the serial monitor connection. Both showed the problem when using PlatformIO, but not when using VS Code to work with ESP-IDF. (Perhaps deactivating the autoclose feature might help for the JTAG case, but I haven't tried it.)

foofel, 2022-10-09:
exact same behaviour here with PlatformIO 6.1.4

Panometric, 2022-11-13:
Can't believe this is still open in 2022. Is there something platform specific? @pearson observation that it works sometimes suggests a race condition.  I've gone back to esp-idf extension because of it. It never has this problem.