platformio/platformio-vscode-ide #1957: No Module Named Package

hzhen95, 2020-08-25:
I came across this problem and i cant find ways to solve it 

> Executing task: C:\Users\HouZhen\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1768; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
ImportError: No module named package:
  File "C:\Users\HouZhen\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 167:
    env.SConscript(item, exports="env")
  File "C:\Users\HouZhen\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\HouZhen\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\HouZhen\Desktop\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 12:
    from platformio.managers.package import PackageManager

Seems like package is not in platformio.manager
I have tried to upgrade platformio 
Thanks

sl1pkn07, 2020-09-04:
update your marlin (seems fixed in bugfix-2.0.x branch)

ivankravets, 2020-10-16:
Should be fixed in PlatformIO IDE 2.0 for VSCode.

closed 2020-10-16 by ivankravets