platformio/platformio-vscode-ide #1960: Unable to build sketch: \platformio.exe” does not exist

NitrofMtl, 2020-08-25:
Hi.
I recently reinstall my windows, then PlatformIO.
The first time I’ve done it, it work out of the box. This time I am unable to build my sketch. Console return:

    Executing task in folder 200812-053037-due: c:\Users\[myuser]\OneDrive\Documents\PlatformIO\Projects\200812-053037-due\platformio.exe run <

    The terminal process failed to launch: Path to shell executable "c:\Users\<[myuser]\OneDrive\Documents\PlatformIO\Projects\200812-053037-due\platformio.exe" does not exist.

I tried with import a project and create a new one with the same resutl.

win10 home, 120.2212.31.0
vs Code 1.48.0
PlatformIO core 4.3.4 home 3.2.3
Board Arduino DUE

I've [post first on community](https://community.platformio.org/t/unable-to-build-sketch-platformio-exe-does-not-exist/15551) ,install core, and it compile into core, but the bug persist on vs code.

NitrofMtl, 2020-09-25:
seem fix on home 3.3.0, core 5.0.1

closed 2020-09-25 by NitrofMtl

poikilos, 2022-09-27:
TLDR I installed the new cross-platform PowerShell and that fixed the issue for me.

Someone suggested PowerShell was the issue at https://community.platformio.org/t/unable-to-build-sketch-platformio-exe-does-not-exist/15551/22 and suggested switching PlatformIO to use Command Prompt but I thought that was overkill so I upgraded PowerShell as VSCode suggested (from https://github.com/PowerShell/PowerShell/releases/tag/v7.2.6). I closed VSCode, installed the upgraded Win64 MSI, then re-opened VSCode. After a minute or so, I started automatically getting info in the "Output" tab saying that PlatformIO Core was installing. After that, PlatformIO worked like it did when I tried it last year or so.

NitrofMtl, 2022-09-27:
I don't know exactly what the issue is. For me, the solution was to not initialiaze project into powershell but with the PIO GUI.
I did not try init using vscode command promt.