platformio/platformio-vscode-ide #1963: Start PIO Home Server

kinggrave, 2020-08-26:
# Description of problemLeave a comment...BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN- https://github.com/platformio/platformio-vscode-ide/issues# ConfigurationVSCode: 1.48.2PIO IDE: v1.10.0System: Windows_NT, 10.0.18363, x64# Exception```Error: Error: Traceback (most recent call last):File "c:\users\matthias\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in maincli()  # pylint: disable=no-value-for-parameterFile "c:\users\matthias\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__return self.main(*args, **kwargs)File "c:\users\matthias\.platformio\penv\lib\site-packages\click\core.py", line 782, in mainrv = self.invoke(ctx)File "c:\users\matthias\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invokereturn super(PlatformioCLI, self).invoke(ctx)File "c:\users\matthias\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invokereturn _process_result(sub_ctx.command.invoke(sub_ctx))File "c:\users\matthias\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invokereturn ctx.invoke(self.callback, **ctx.params)File "c:\users\matthias\.platformio\penv\lib\site-packages\click\core.py", line 610, in invokereturn callback(*args, **kwargs)File "c:\users\matthias\.platformio\penv\lib\site-packages\platformio\commands\home\command.py", line 54, in clifrom autobahn.twisted.resource import WebSocketResourceFile "C:\Users\Matthias\.platformio\packages\contrib-pysite\autobahn\twisted\__init__.py", line 40, in <module>from autobahn.twisted.util import sleepFile "C:\Users\Matthias\.platformio\packages\contrib-pysite\autobahn\twisted\util.py", line 31, in <module>from twisted.internet.defer import DeferredFile "C:\Users\Matthias\.platformio\packages\contrib-pysite\twisted\internet\defer.py", line 32, in <module>from twisted.logger import LoggerFile "C:\Users\Matthias\.platformio\packages\contrib-pysite\twisted\logger\__init__.py", line 102, in <module>from ._logger import Logger, _loggerForFile "C:\Users\Matthias\.platformio\packages\contrib-pysite\twisted\logger\_logger.py", line 274, in <module>_log = Logger()File "C:\Users\Matthias\.platformio\packages\contrib-pysite\twisted\logger\_logger.py", line 70, in __init__from ._global import globalLogPublisherFile "C:\Users\Matthias\.platformio\packages\contrib-pysite\twisted\logger\_global.py", line 16, in <module>from ._buffer import LimitedHistoryLogObserverFile "C:\Users\Matthias\.platformio\packages\contrib-pysite\twisted\logger\_buffer.py", line 11, in <module>from zope.interface import implementerModuleNotFoundError: No module named 'zope'============================================================An unexpected error occurred. Further steps:* Verify that you have the latest version of PlatformIO using`pip install -U platformio` command* Try to find answer in FAQ Troubleshooting sectionhttps://docs.platformio.org/page/faq.html* Report this problem to the developershttps://github.com/platformio/platformio-core/issues============================================================at c:\Users\Matthias\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:11965at ChildProcess.d (c:\Users\Matthias\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:8215)at ChildProcess.emit (events.js:203:13)at ChildProcess.cp.emit (c:\Users\Matthias\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)at maybeClose (internal/child_process.js:1021:16)at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)```

kinggrave, 2020-08-26:
Please help me fix this Problem .. THX

HeartlandAgTech, 2020-08-31:
I figured out a similar issue.  With the release of 1.10.0, the PIO core needed to be reinstalled for me.  

The note in this issue helped resolve the problem:  https://github.com/platformio/platformio-vscode-ide/issues/1353

"Please remove HOME_DIR/.platformio/penv folder after update to 1.10.0 and restart VSCode."

closed 2020-10-07 by stale[bot]