platformio/platformio-vscode-ide #1990: Installation Manager

kirk-loeten, 2020-09-05:
I would like to link python on MACos. /Applications/Python 3.8/IDLE.app

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.48.2
  PIO IDE: v1.10.0
  System: Darwin, 18.7.0, x64

  %23 Exception
  ```
  Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: TypeError [ERR_INVALID_ARG_TYPE]: The "file" argument must be of type string. Received type undefined

	at t.value (/Users/Chris/.vscode/extensions/platformio.platformio-ide-1.10.0/node_modules/platformio-node-helpers/dist/index.js:1:34077)
	at processTicksAndRejections (internal/process/task_queues.js:85:5)
  ```

oldfellr, 2020-09-06:
Any work around for this?  I'm not a programmer at all!

ivankravets, 2020-09-11:
Please upgrade VSCode and PlatformIO IDE to the latest versions. Should be fixed.

closed 2020-09-11 by ivankravets