platformio/platformio-vscode-ide #1995: Administator permission required to delete .pio folder after a failed build

Bmooij, 2020-09-07:
When a build fails during installing the dependencies, it requires Administator privilage to delete the .pio folder.
Only when I close VS code, I can properly remove the .pio folder. 
(The .pio folder needs to be deleted, otherwise it doesn't download the dependencies again)

PlatformIO Core: 5.0.0
PlatformIO Home: 3.3.0
PlatformIO IDE: 1.10.0
Pyhton: 3.7.9

My platformio.ini
```ini
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#v1.12.4
board = esp32dev
framework = arduino
lib_deps =
    ESP Async WebServer = https://github.com/Bmooij/ESPAsyncWebServer.git#dc2546c
```

Bmooij, 2020-09-09:
Root cause found!

The problem is that vs code automatically watches for the .git folder. It locks ".git" folders that are present in the dependency folder. 
When closing the vscode, this folder is unlocked and can be removed.

It also explains `[WinError 5] Access is denied` errors that references to `PROJECTLIBDEPS_DIR\.pio\libdeps\ENV\PACKAGE\.git\objects\**` 

To fix this issue I completely disabled the git integration in vscode.
`settings.json` I added:
```json
{
    "git.enabled": false
}
```

mcspr, 2020-09-09:
Meaning, vscode git extension does not respect .gitignore?
(or, it interprets it differently / works with git commands in some strange way)

ref. https://github.com/xoseperez/espurna/blob/f6ec2422a49f303658cd2d96fb6c3d14cab861a1/code/.gitignore#L12
```
.pio
```
(from platformio/platformio-core#3328)

Bmooij, 2020-09-09:
Nope, see: https://github.com/microsoft/vscode/issues/62725

ivankravets, 2020-10-16:
Could you reproduce this issue with `pio upgrade --dev`?

closed 2020-12-22 by stale[bot]