platformio/platformio-vscode-ide #1999: Start PIO Home Server

chadox, 2020-09-08:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.48.2
  PIO IDE: v1.10.0
  System: Linux, 4.9.0-13-amd64, x64

  %23 Exception
  ```
  Error: Error: You are not connected to the Internet.
PlatformIO needs the Internet connection to download dependent packages or to work with PlatformIO Account.

	at /home/ahz/.vscode/extensions/platformio.platformio-ide-1.10.0/node_modules/platformio-node-helpers/dist/index.js:1:11965
	at ChildProcess.d (/home/ahz/.vscode/extensions/platformio.platformio-ide-1.10.0/node_modules/platformio-node-helpers/dist/index.js:1:8215)
	at ChildProcess.emit (events.js:203:13)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

chadox, 2020-09-08:
Hi
Some trouble about proxy ?
Thanks

ivankravets, 2020-10-16:
Should be fixed in PlatformIO IDE 2.0 for VSCode.

closed 2020-10-16 by ivankravets