platformio/platformio-vscode-ide #2002: [FR] Ability to load .env files for dynamic variables

PhilippMolitor, 2020-09-10:
# Introduction

Examine this example case:

### platformio.ini
```ini
[env]
platform = espressif32
board = esp32dev
framework = arduino
build_flags =
  -D WIFI_SSID=\"${sysenv.WIFI_SSID}\"
  -D WIFI_PASS=\"${sysenv.WIFI_PASS}\"
```

In this case, I am trying to add my WiFi credentials to the code. An important aspect for me is to NOT have those in my git repository, so I use the environmental config.

# Issue

While this being a viable option when using the PlatformIO CLI, where I can load and set environment variables on the fly, using this configuration with the VS Code IDE Plugin does not work for me. I didn't find any configuration that allows me to load environmental variables such as other projects often do:

[https://pypi.org/project/python-dotenv/](https://pypi.org/project/python-dotenv/)
[https://www.npmjs.com/package/dotenv](https://www.npmjs.com/package/dotenv)

This way, I can still locally configure my secrets while keeping them out of the git repository / away from public.

# Example solution

Adding a configuration option for environments:

```ini
[env]
...

dotenv_files =
  .env.common

...

[env:release]
dotenv_files =
  {env.dotenv_files}
  .env.production
```

Where all settings common in both files will be overridden by `.env.production` in this case as it is listed last.

ivankravets, 2020-09-10:
Does it work with the latest  PlatformIO IDE 2.0? https://community.platformio.org/t/platformio-ide-for-vscode-2-0/15893

PhilippMolitor, 2020-09-10:
I'm not sure this is related to the 2.0 release? I opened a feature request as I could not find any matching feature in the docs.

ivankravets, 2020-09-10:
Please update to the latest version. Does it work now?

PhilippMolitor, 2020-09-10:
I already am on version 2.0. I'm not sure if you are understanding my request.
The option "dotenv_files" does not exist, and this is a feature request.

PhilippMolitor, 2020-09-10:
I created an intermediate solution, which does the trick temporarily, but is far from optimal:

platformio.ini
```ini
[platformio]
extra_configs = 
  secrets.ini
build_flags =
  -D WIFI_SSID=\"${env:secrets.wifi_ssid}\"
  -D WIFI_PASS=\"${env:secrets.wifi_pass}\"
```

secrets.init
```
[env:secrets]

wifi_ssid = MyWiFiSSID
wifi_pass = MyWiFiPassword
```

The following warnings occurr:
```
Warning! Ignore unknown configuration option `wifi_ssid` in section [env:secrets]
Warning! Ignore unknown configuration option `wifi_pass` in section [env:secrets]
```

Also, this does not solve the initial problem of not being able to include environment variables from a file.

ivankravets, 2020-09-11:
Docs https://docs.platformio.org/en/latest/projectconf/section_platformio.html#extra-configs

closed 2020-09-11 by ivankravets

PhilippMolitor, 2020-09-11:
Do you even read what I write here? This was a sketchy workaround, not a fix, and certainly not a feature implementation of the **feature request**.

ivankravets, 2020-09-11:
Everything that you need is already implemented. You can decide how to load your envs and convert the to build flags. If you don't like "extra_configs" approach, please take a look at https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html

Please use the extra script and the power of Python language.