platformio/platformio-vscode-ide #2004: Broken `Terminal` menu

puzrin, 2020-09-10:
In 2.0.0 / 2.0.1

1. `Terminal` -> `Run Build Task` - no env-related items displayed (only single one with default target).
2. `Terminal` -> `Run Task` - the same.

I need a very simple thing - build/upload for selected hardware. Now this can be done only via explorer - too difficult to reach.

How to reproduce:

- Open https://github.com/speedcontrols/ac_sc_grinder
- Try `Terminal` -> `Run Task` / `Run Build Task`
- Go to extensions and force install 1.10.0, then compare content of menus.

ivankravets, 2020-09-14:
Fetching ALL envs is expensive operation. PlatformIO IDE 2.0 now uses PlatformIO Core 5.0 that provide dynamics targets. See https://docs.platformio.org/en/latest/core/migration.html#custom-targets

We can't preload all targets from ALL envs by default. It does not make sense. Now, the developer personally decide which tasks to load. So, if you expand env tasks via `PlatformIO: Tasks` view - you will see them in `Run Build Task`.

Please note that you can drag `PlatformIO: Tasks` into the "Explorer" activity to avoid switching between activities.

puzrin, 2020-09-15:
I don't agree with this approach:

- First, code is written for humans. It's not correct to argument with "feature dropped because implementaion become difficult".
- I do not need ALL possible targets preload into menu, only frequently used (build/upload, no custom)
- Your proposal about explored leafs expand could work, but:
  - PIO does not remember state after VSCode reload.
  - No way to define default state on first load.

---

Please, reconsider your decision. I have concrete use case: pre-populate `Terminal` menu with desired content (for me - 2 existing targets for each env). IMO such kind of use cases can be implemented somehow, with low CPU cost.

ivankravets, 2020-09-15:
Thanks, I renamed this issue. We will add a new setting that will instruct PlatformIO to pre-load ALL tasks/targets.

closed 2020-09-16 by ivankravets

ivankravets, 2020-09-16:
Please update to PlatformIO IDE 2.1.0 for VSCode. See https://github.com/platformio/platformio-vscode-ide/releases/tag/v2.1.0

puzrin, 2020-09-16:
Much better, thanks!

Is it possible to force this on project open (via `/.vscode` folder or something else)? I mean use case, when newbie (not developer) open project and wish to upload firmware.

ivankravets, 2020-09-16:
> I mean use case, when newbie (not developer) open project and wish to upload firmware.

Do you mean that it does not work now?

puzrin, 2020-09-16:
It works. But it expects user know how to change settings. That's normal for developpers, but not normal for ordinary users.

See [this project](https://github.com/speedcontrols/ac_sc_grinder) for example. End users are very far from programming. They follow instruction like [this](https://github.com/speedcontrols/ac_sc_grinder/blob/master/doc/assembly.md#upload-firmware-and-test).

ivankravets, 2020-09-16:
Sorry, I still do not understand the problem. 

> Run via VS Code menu: Terminal → Run Task... → PlatformIO: Upload, and wait until complete.

I've just checked and it works.

puzrin, 2020-09-16:
> Sorry, I still do not understand the problem.

This feature (platformio-ide.autoPreloadEnvTasks) is disabled by default, and user should have appropriate "skills" to enable.

This may sound stupid, but this is a problem for users, who install vscode only to upload firmware. Developpers are ok.

ivankravets, 2020-09-16:
We can't do this by default. There are projects with dozen envs. See:
- https://github.com/MarlinFirmware/Marlin
- https://github.com/arendst/Tasmota

Could you make a PR to that repositories and ask maintainers to follow our official documentation https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks?

We are working on the updates to that docs to reflect all new features from PlatformIO IDE 2.0.

puzrin, 2020-09-16:
> We can't do this by default. There are projects with dozen envs. See:

I did no asked to change defaults. I asked if this can be "forced" from local project. For example, with [such](https://github.com/speedcontrols/ac_sc_grinder/blob/master/.vscode/extensions.json) file vscode suggests to install pio if not exists. May be something possible for seetting state?

> Could you make a PR to that repositories and ask maintainers to follow our official documentation [docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks](https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks)?

PR is not a problem. Problem is breaking stable processes & previous releases :). This should not depend on IDE.

We already had incedent with broken releases, when older `etl` versions disappeared from registry. Now reason is different, but result similar. I try to understand, if this can be dimmed somehow. In nothing more can be done - ok.

PS. Anyway, thanks for 2.1.0 updates. This quick-fixed annoying problem personally for me.

ivankravets, 2020-09-17:
Could you open a separate issue? Maybe it makes sense to preload all env tasks if the project contains < 5?10? envs. Or, preload ALL and allow people to DISABLE this.

To be honest, there is no technical issue. We just think about users and the performance of this extension.

ivankravets, 2020-09-17:
Thanks for the #2018!

>  i will not use pio anymore

Is it linked to this issue? :( Or, there are other issues with PlatformIO? Sorry.

puzrin, 2020-09-17:
> Is it linked to this issue? :( Or, there are other issues with PlatformIO? Sorry.

The only issue is my bad english :). I mean, i'm NOT person who say "i will not use your product if bug not fixed".

ivankravets, 2020-09-19:
Ah, sorry, I thought you will not use PlatformIO anymore. I think we can enable #2018 by default in the next release and gather feedback.

commit a2705967 (2020-09-16): Added a new setting ``platformio-ide.autoPreloadEnvTasks`` to enable automatic preloading of the project environment tasks // Resolve #2004