platformio/platformio-vscode-ide #2007: PlatformIO break

Gizmodo, 2020-09-11:
Version: 1.50.0-insider
Commit: bc38b0accb3c62180317c63dbe7323200977db0f
Date: 2020-09-11T12:26:02.664Z
Electron: 9.3.0
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Darwin x64 19.6.0

PlatformIO shows me this error:
[platformio.platformio-ide]: Proposed API is only available when running out of dev or with the following command line switch: --enable-proposed-api platformio.platformio-ide
Reinstall dont work

ivankravets, 2020-09-11:
Does it work with the stable VSCode? Not insider build.

skieast, 2020-09-12:
Same issue here with VScode Insiders
I can run commands from the sidebar but not the command palette. 
Uninstalled and reinstalled same issue.  

Version: 1.50.0-insider (user setup)
Commit: cca20eba8974b1d6b6a9b71bf4e60a29c8f26648
Date: 2020-09-12T01:04:21.398Z
Electron: 9.3.0
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.20211

Installed VScode and it works. This was a new install with no other extensions.

ivankravets, 2020-09-12:
Please report to https://github.com/microsoft/vscode/issues

closed 2020-09-12 by ivankravets