platformio/platformio-vscode-ide #2012: Clicking clone git project does nothing

Taubin, 2020-09-13:
```
Version: 1.50.0-insider (user setup)
Commit: cca20eba8974b1d6b6a9b71bf4e60a29c8f26648
Date: 2020-09-12T01:04:21.398Z
Electron: 9.3.0
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19042
```

Reinstalled VSCode as well as PlatformIO entirely however when I click on "clone git project" nothing happens. There are no errors at all, and the popup does not come up. 

I've upgraded platformio using the upgrade button and it states there are no updates
```

platformio upgrade
You're up-to-date!
PlatformIO 5.0.1 is currently the newest version available.
```

I'm not sure what I can do to assist further, please let me know what else is needed.

callsSolve, 2020-09-14:
platformio upgrade
You're up-to-date!
PlatformIO 5.0.1 is currently the newest version available.

ivankravets, 2020-09-14:
>  I click on "clone git project" nothing happens.

It seems that your question is related to the VSCode. Please check that you installed Git client https://git-scm.com/downloads

closed 2020-09-14 by ivankravets