platformio/platformio-vscode-ide #2014: Issue with built-in Python

maakbaas, 2020-09-15:
I have received at least three reports of people using the [ESP8266 IoT Framework](https://github.com/maakbaas/esp8266-iot-framework) where updating PlatformIO has led to the build errors below.

The issue seems to be resolved by manually uninstalling python and installing 3.7.7 instead of 3.8.x. Searching on Google for these errors always seems to lead to problems where Python versions are mixed.

If you look at for instance [this issue](https://github.com/maakbaas/esp8266-iot-framework/issues/36) it seems that disabling the new option for using built-in python does not seem to change anything.

Any explanation or solution for this issue would be appreciated!

```<SCons.Script.SConscript.SConsEnvironment object at 0x00000133930F0C48>
Traceback (most recent call last):
File "C:\Users\xxx\Documents\PlatformIO\Projects\yyy.pio\libdeps\d1_mini_pro\ESP8266 IoT Framework\build\preBuildConfig.py", line 4, in
import inspect, os.path
File "C:\Users\xxx.platformio\python3\lib\inspect.py", line 38, in
import importlib.machinery
File "C:\Users\xxx.platformio\python3\lib\importlib_init_.py", line 51, in
_w_long = _bootstrap_external._w_long
AttributeError: module 'importlib._bootstrap_external' has no attribute '_w_long'
Traceback (most recent call last):
File "C:\Users\xxx\Documents\PlatformIO\Projects\ifttt-d1_mini_pro-sleep_v2.pio\libdeps\d1_mini_pro\ESP8266 IoT Framework\build\preBuildCertificates.py", line 7, in
from asn1crypto.x509 import Certificate
File "C:\Users\xxx\AppData\Local\Programs\Python\Python38-32\lib\site-packages\asn1crypto\x509.py", line 19, in
from encodings import idna # noqa
File "C:\Users\xxx.platformio\python3\lib\encodings\idna.py", line 3, in
import stringprep, re, codecs
File "C:\Users\xxx.platformio\python3\lib\stringprep.py", line 8, in
from unicodedata import ucd_3_2_0 as unicodedata
ImportError: DLL load failed while importing unicodedata: %1 is not a valid Win32 application.
Found 33 compatible libraries```

ivankravets, 2020-09-16:
The latest PlatformIO IDE 2.0 for VSCode comes with built-in Python (we have just renamed it to "Use a portable Python 3 Interpreter if available""). We use pre-bilt Python 3.7.7.

So, please use our portable Python 3.

maakbaas, 2020-09-16:
I appreciate your response, but I feel that this answer is too simple. 

I am ok with PlatformIO using an internal Python version, but right now the implementation is such that simply having another Python version installed on your PC (for other stuff besides PlatformIO) will cause issues?

ivankravets, 2020-09-16:
> C:\Users\xxx\AppData\Local\Programs\Python\Python38-32

It looks like our portable Python was ignored. Could you remove `C:\Users\xxx.platformio\penv` folder and provide more details opening VSCode > Help > Developer Tools > Console?

maakbaas, 2020-09-16:
I have resolved the issue, and it was 100% on me, so sorry for taking your time.

In my framework I have a script defined in `extra_scripts` in `platformio.ini` that was calling other Python scripts using `subprocess.call`. This is where the mix-up of Python versions happened. I have now changed these child scripts into Python functions to call them directly from within the Python environment, and now the issue is resolved.

ivankravets, 2020-09-16:
Please check our community forums. I’ve recently posted how to correctly call Python from extra scripts. It seems that we should update our docs. I’ve just added this case to my TODO

closed 2020-09-16 by ivankravets

wdmtech, 2020-10-11:
I _think_ this is related - I'm not trying to do anything with certificates, but creating a new project in PlatformIO, adding `lib_deps = ESP8266 IoT Framework` to `platformio.ini`, and then building results in the following:

```
> Executing task in folder esp8266-iot-framework: C:\Users\Will\.platformio\penv\Scripts\platformio.exe run <

Processing esp01_1m (platform: espressif8266; board: esp01_1m; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 (2.6.2) > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing ESP8266 IoT Framework
Library Manager: ESP8266 IoT Framework @ 1.3.2 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing ESP Async WebServer
Library Manager: ESP Async WebServer @ 1.2.3 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing ESPAsyncTCP
Library Manager: ESPAsyncTCP @ 1.2.2 has been installed!
Library Manager: Installing AsyncTCP
Library Manager: AsyncTCP @ 1.1.1 has been installed!
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Library Manager: Installing ArduinoJson
Library Manager: ArduinoJson @ 6.16.1 has been installed!
ModuleNotFoundError: No module named 'asn1crypto':
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Will\.platformio\platforms\espressif8266\builder\main.py", line 238:
    target_elf = env.BuildProgram()
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1064:
    lib_builders = env.GetLibBuilders()
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 993:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 61:
    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 137:
    self.process_extra_options()
  File "C:\Users\Will\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 282:
    exports={"env": self.env, "pio_lib_builder": self},
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Will\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Will\Documents\PlatformIO\Projects\esp8266-iot-framework\.pio\libdeps\esp01_1m\ESP8266 IoT Framework\build\preBuild.py", line 9:
    from preBuildCertificates import preBuildCertificatesFun
  File "C:\Users\Will\Documents\PlatformIO\Projects\esp8266-iot-framework\.pio\libdeps\esp01_1m\ESP8266 IoT Framework\build\preBuildCertificates.py", line 7:
    from asn1crypto.x509 import Certificate
==================================================== [FAILED] Took 5.92 seconds ====================================================
The terminal process "C:\Users\Will\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

```