platformio/platformio-vscode-ide #2015: Start PIO Home Server

ami3go, 2020-09-15:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Windows_NT, 10.0.17763, x64

  %23 Exception
  ```
  Error: Error: HTTPSConnectionPool(host='api.registry.ns1.platformio.org', port=443): Max retries exceeded with url: /v3/packages%3Fquery=type%3A%22tool%22 name%3A%22contrib-piohome%22 (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)')))

	at c:\Users\achestni\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:12033
	at ChildProcess.c (c:\Users\achestni\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\achestni\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-09-16:
> HTTPSConnectionPool(host='api.registry.ns1.platformio.org', port=443): Max retries exceeded with url: /v3/packages%3Fquery=type%3A%22tool%22 name%3A%22contrib-piohome%22 (Caused by [...]

Please check your internet connection.

closed 2020-09-16 by ivankravets

ami3go, 2020-09-17:
If there any was to make it work without internet. Im trying so set platformIO, but probably i have quite strict firewall wall. There is no way for me to desible firewall as i do not have an access to it