platformio/platformio-vscode-ide #2018: Preload some tasks by default

puzrin, 2020-09-17:
In continue of https://github.com/platformio/platformio-vscode-ide/issues/2004#issuecomment-694213164

v2 no longer preload targets to `Terminal-> Run Build Task` & `Terminal -> Run Task` by default. 

- This may be good for large projects with multiple envs (works faster)
- This is not convenient for people who prefer run tasks via menus
- There is `platformio-ide.autoPreloadEnvTasks` option, but it's global - sill need switch when load different projects.

It would be nice to allow developpers customize preloaded targets somehow.

----

Ideas below are incomplete, and not ideal. Only to start brainstorming.

- Preload all by default, and global option to disable (reverse current behaviour)
  - still no customizeable for different projects.
- Override settings via `/.vscode/...` (i don't know how, but should be possible)
- In `plaform.ini`, for each env `preload_ide_targets = [ foo, bar ]` or `*`
  - probably, no best place to customize IDE
- [your ideas]

What is desired, in ideal world:

- Allow customise defaults via project configs [project level setting]
- Still allow everyone to override customized defauls, accordng to own preferences, without need to commit changes [user level setting for concrete project].

ivankravets, 2020-10-23:
I think the good solution for this issue is to ALWAYS preload tasks from an active working environment (you can switch from the bottom bar).

Also, would be great to SHOW PLATFORM TASKS on the TOP level in "Project Task Explorer. See https://www.reddit.com/r/esp32/comments/jgncc4/thoughts_on_platformio/g9rra1p?utm_source=share&utm_medium=web2x&context=3

puzrin, 2020-10-24:
> I think the good solution for this issue is to ALWAYS preload tasks from an active working environment (you can switch from the bottom bar).

IMO you missundersand use case. I have this ones (may be more exists):

- Developper: random exec of run/build/test in different envs
- End user: reach fimware upload command

For both cases, going to bottom bar env switch is addional loosing of view context.

Also, pay attention, in many cases developper is NOT focused on concrete env.

---

I suggest to return "load everything" in some configurable form. I can accept fact, that you may not wish do that by default. But if this could be configured by project author for everybody who cloned and opened project first time - fine.

pfeerick, 2020-10-28:
> But if this could be configured by project author for everybody who cloned and opened project first time - fine.
In that case, in the `platformio.ini` under the `[platformio]` would be the best place for a such a preload setting - since it is this file that contains all the project specific settings. There you could list either list one, many or wildcard the envs to preload.

puzrin, 2020-10-28:
@pfeerick that was mentioned in first post, with comment "`platformio.ini` is not correct place to store IDE-specific settings". If this can be done via `.vscode/` folder, that would be ideal.

Problem with `.vscode/` is - user's local changes should not be added to `git commit`.

pfeerick, 2020-10-28:
There's never going to be a simple solution that satisfies both. As you point out, `.vscode/` should not be synced because that is the local VSCode specific config. That does however leave room for 'pre-load by default', and 'allow local customisation'... , if that is the desired behaviour. However, I also see it similar to environments, where you might only be interested in a couple, or one, or all... hence the `default_envs` parameter... hm... perhaps it can be heirachical... pre-load all by default (can be disabled by global setting), otherwise, pre-load those specified in default_envs, else local .vscode override...

puzrin, 2020-10-28:
https://github.com/speedcontrols/ac_sc_grinder/tree/master/.vscode - i use partial `.vscode/` sync, to auto-install pio on project open. That's important for users who just need firmware upload. Parial sync works.

Anyway, 2 level of customisation required:

- Change default behaviour (for all users of concrete project). Should be commit-able to repo.
- Change local user behaviour for concrete project. Should not be commit-able to repo.

ivankravets, 2020-10-31:
Hi all,

We did a lot of improvements in the upcoming PlatformIO IDE 2.2.0 for VSCode. Could you try it? Does it solve this issue?

* Changes => https://github.com/platformio/platformio-vscode-ide/blob/develop/CHANGELOG.md
* VSIX => https://www.dropbox.com/s/ybmiw2of974bykg/platformio-ide-2.2.0-beta.1.vsix?dl=0

[image]

puzrin, 2020-10-31:
> We did a lot of improvements in the upcoming PlatformIO IDE 2.2.0 for VSCode.

Improvments are nice.

> Could you try it? Does it solve this issue?

No. Changes are not related to described problem. https://github.com/platformio/platformio-vscode-ide/issues/2018#issuecomment-717832035