platformio/platformio-vscode-ide #2023: Python libs build for Mac OS X 10.13 do not worn on previous version os x

hackebike, 2020-09-23:
Processing BIGTREE_GTR_V1_0 (platform: ststm32@>=5.7.0,<6.2.0; board: BigTree_GTR_v1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing ststm32 @ >=5.7.0,<6.2.0
Downloading  [####################################]  100%          
dyld: lazy symbol binding failed: Symbol not found: _utimensat
  Referenced from: /Users/biskero/.platformio/penv/bin/python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _utimensat
  Referenced from: /Users/biskero/.platformio/penv/bin/python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib

Terminal will be reused by tasks, press any key to close it.

ivankravets, 2020-10-16:
Do you use the latest PlatformIO IDE?

hackebike, 2020-10-16:
Yes,
Platformio Core 5.0.1·Home 3.3.0

basically yhe python build is been compiled for macos 10.13 but I am on 10.12.6.
It was working fine until few weeks ago.

ivankravets, 2020-10-16:
The version of extension.

hackebike, 2020-10-16:
Sorry, platformio v2.1.0

ivankravets, 2020-10-16:
Please install the latest Python 3.8 (not 3.9) using this instruction https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

Then, open VSCode settings and search for `platformio builtin python`, uncheck it. Restart VSCode.

closed 2020-10-16 by ivankravets

hackebike, 2020-10-16:
ok I installed Python 3.8.6 for macOS, did the uncheck, restarted but getting the same error:

> Executing task: platformio run <

Processing BIGTREE_GTR_V1_0 (platform: ststm32@~8.0; board: BigTree_GTR_v1; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing ststm32 @ ~8.0
Downloading  [####################################]  100%
dyld: lazy symbol binding failed: Symbol not found: _utimensat
  Referenced from: /Users/biskero/.platformio/penv/bin/python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _utimensat
  Referenced from: /Users/biskero/.platformio/penv/bin/python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib

Terminal will be reused by tasks, press any key to close it.

hackebike, 2020-10-16:
so if I do which python
/usr/bin/python
and it's version 2.7, event though I just installed 3.8.6.
Seems it's not been updated or the path to the new Python is not been set correctly.

ivankravets, 2020-10-16:
It's good. There is `python` and `python3`.

hackebike, 2020-10-16:
ok but the VSC does not see the new installed python and continues to use /Users/biskero/.platformio/penv/bin/python even though I uncheck Use Builtin Python
So not sure how to fix this

hackebike, 2020-10-28:
The problem is still present and the "solutions" suggested did not work. Anyone can fix this?

Platform Manager: Installing ststm32 @ ~8.0
Downloading  [####################################]  100%          
dyld: lazy symbol binding failed: Symbol not found: _utimensat
  Referenced from: /Users/biskero/.platformio/penv/bin/python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _utimensat
  Referenced from: /Users/biskero/.platformio/penv/bin/python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib

Terminal will be reused by tasks, press any key to close it.