platformio/platformio-vscode-ide #2032: Unable to install platformio-ide extension - Extension not listed

hafu, 2020-09-26:
On a fresh install of VS Code (OSS) I am unable to install the platformio-ide extension. It is not listed in the search or running the commands [described in the documentation](https://docs.platformio.org/en/latest/integration/ide/vscode.html#installation).

[image]

On another machine, where it is already installed the messages `ModuleNotFoundError: No module named 'platformio.managers.package'`. Maybe they are related.

---

VS Code OSS:
```
Version: 1.49.1
Commit: 58bb7b2331731bf72587010e943852e13e6fd3cf
Date: 2020-09-23T22:26:38.399Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Linux x64 5.8.10-arch1-1
```

abu-noor, 2020-09-27:
Installing visual-studio-code-bin(AUR) (Microsoft-branded release), shows PlatformIO and others. Really didn't want to install the branded one. I could install the extensions manually, but I'm getting lazy in my later years.

ivankravets, 2020-09-27:
Just search for "platformio".

closed 2020-09-27 by ivankravets

abu-noor, 2020-09-27:
@ivankravets Arch linux's community vscode(code), doesn't show all extensions in the search, platformio being one of them. You have to install the AUR branded ones.

ivankravets, 2020-09-27:
If this is not official VSCode, please use this instruction https://github.com/platformio/platformio-vscode-ide/issues/1802

hafu, 2020-09-28:
Thanks a lot. 

I was confused, because some weeks before it worked without problems. I looked into the changes of the arch package and found the cause, they changed the markedplace url: https://github.com/archlinux/svntogit-community/commit/b62a02ec645f16528c0269faf12e362a55efda75?branch=b62a02ec645f16528c0269faf12e362a55efda75&diff=split

Will go for the temporarily solution.

etienne1911, 2020-10-11:
Hi 
I'm also having this issue on Arch.
@hafu do you know why they made this change and if it will be reverted?
did you install outside marketplace?

hafu, 2020-10-12:
@etienne1911 I don't know why this change was made and I think it won't be reverted. I assume some license issues. There are the two possible options:

1. User the workaround (manual installation) mentioned in #1802 or steps in my [initial issue](https://github.com/platformio/platformio-vscode-ide/issues/2032#issue-709519737)
2 . Or use the [AUR package visual-studio-code-bin](https://aur.archlinux.org/packages/visual-studio-code-bin/) mentioned in the [comment](https://github.com/platformio/platformio-vscode-ide/issues/2032#issuecomment-699666707) 

It looks like it will took some time until PlatformIO is available in OpenVSX (which is used by code oss). PlatformIO have some dependencies, which have restrictive licenses. See: https://github.com/open-vsx/publish-extensions/issues/86

Fireflaker, 2025-01-10:
• [Download vsix from vsixhub server](https://www.vsixhub.com/go.php?post_id=1912&app_id=6cf182b2-70cf-433e-97f2-45f414c9ad89&s=v5ixzAiKpRBwk&link=https%3A%2F%2Ff2.vsixhub.com%2Ffile.php%3Fversion%3D3.3.3%26ext_name%3Dplatformio-ide)

• [Download vsix from rapidgator.net](https://www.vsixhub.com/go.php?post_id=1912&app_id=6cf182b2-70cf-433e-97f2-45f414c9ad89&s=v5ixzAiKpRBwk&link=https%3A%2F%2Frapidgator.net%2Ffile%2Fd3d7317f4511c8a94884561c6dc6b883%2F)