platformio/platformio-vscode-ide #2038: [Feature] Add option to skip PIO installation.

cab404, 2020-09-27:
I use nix shell to provide my own version of PlatformIO — and it's quite frustruating to cancel installation each time VSCode starts up.

I can't use bundled methods of installation, cause NixOS.

Option to disable installation and use `pio` from PATH would be great.

pak-man, 2021-09-09:
I second this. Would be so so ergonomic / useful. Even without custom icon just a short label + tooltip on mouseover would be good enough.

ivankravets, 2022-07-23:
See IDE's settings https://docs.platformio.org/en/latest/integration/ide/vscode.html#settings

I use my custom PlatformIO Core with the next configuration:

```json
{
    "platformio-ide.customPATH": "~/Projects/PlatformIO/Development/core/.tox/py39/bin",
    "platformio-ide.useBuiltinPIOCore": false,
    "platformio-ide.useBuiltinPython": false,
}
```

closed 2022-07-23 by ivankravets