platformio/platformio-vscode-ide #2045: PlatformIO v2.x update breaks the Particle Workbench extension on Windows...

busticated, 2020-10-01:
hello from Particle :wave:

we've recently received a handful of reports ([1](https://community.particle.io/t/local-compile-after-windows-update-to-2004-not-working-make-command-not-found/57716/2), [2](https://community.particle.io/t/issue-workbench-cant-local-compile-due-to-make-command-not-found-on-windows-10/57837)) that our Particle Workbench extension ([link](https://marketplace.visualstudio.com/items?itemName=particle.particle-vscode-pack)) does not play well with the PlatformIO extension ([link](https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide)). notably, this appears to only impact Windows machines.

it seems something in the `v2.0.0` or `v2.0.1` releases ([1](https://github.com/platformio/platformio-vscode-ide/releases/tag/v2.0.0), [2](https://github.com/platformio/platformio-vscode-ide/releases/tag/v2.0.1)) interferes with Workbench's ability to configure the user's local environment.

**Repro Steps**
1. Disable all extensions in VSCode
2. Install Particle Workbench pack extension ([link](https://marketplace.visualstudio.com/items?itemName=particle.particle-vscode-pack))
3. When prompted, allow Workbench to install dependencies
4. Wait for dependencies complete installing
5. Open the VSCode Command Pallet (cmd+shift+p / ctrl+shift+p)
6. Find and run the `Particle: Create Project` command and follow the prompts
7. Wait for your newly created project to load
8. Open the `src/<project-name>.ino` file
9. Find and run the `Particle: Compile application (local)` command
10. Verify project is compiled successfully
11. Install the PlatformIO extension ([link](https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide))
12. Close and reopen VSCode, reloading the project folder you created in step 6 if need be
13. Try to recompile the Particle project via the `Particle: Compile application (local)` command

**Expected**
Project compiles successfully

**Actual**
Project fails to compile, error message of `/usr/bin/bash: make: command not found` is displayed in the active VSCode terminal panel

_if you then disable the PlatformIO extension, click the "reload required" button, and re-try the `Particle: Compile application (local)` command, compilation should succeed again._

**Environment Info**
* os: windows `v10.0.19041` ([info](https://en.wikipedia.org/wiki/Windows_10_version_history#Version_2004_.28May_2020_Update.29))
* arch: x64
* vscode: `v1.49.2`

- - - -

Under the hood, Particle Workbench uses VSCode's `TaskProvider` APIs ([docs](https://code.visualstudio.com/api/references/vscode-api#TaskProvider)) to temporarily configure the user's local environment - specifically prepending paths to their `$PATH` variable and setting a number of environment variables (e.g `PARTICLE_CLI_PATH`, `DEVICE_OS_PATH`, `PLATFORM_ID`, etc) within the task's shell.

To see this in action, you can run the `Particle: Launch Compiler Shell` command and in the terminal panel that launches execute `echo $PATH`. Comparing between the two scenarios, i see:

**With PlatformIO disabled**

41 entries
Particle entries:

```
/cygdrive/c/Users/me/.vscode/extensions/particle.particle-vscode-core-1.12.3/src/cli/bin/windows/amd64
/cygdrive/c/Users/me/.particle/toolchains/gcc-arm/5.3.1/bin
/cygdrive/c/Users/me/.particle/toolchains/openocd/0.11.2-adhoc6ea4372.0/bin
```

**With PlatformIO enabled**

77 entries, many duplicated
no Particle entries

Happy to help troubleshoot further or otherwise work to coordinate things such that our extensions can peacefully coexist :pray::heart::+1:

ivankravets, 2020-10-02:
Hi,

Thanks for the report! We will investigate this issue. 

A temporary solution is to enable this setting https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-activateonlyonplatformioproject

So, PlatformIO's extension will be activated only on PlatformIO's project.

busticated, 2020-10-02:
@ivankravets 

>Thanks for the report! We will investigate this issue.

sure thing! thanks for taking a look :pray:

>A temporary solution is to enable this setting https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-activateonlyonplatformioproject

with that setting enabled within VSCode (screenshot below), i'm still seeing the reported issue. my steps to test were:

1. Open my test project in VSCode
2. Check the `Activate Only On Platform IOProject` option
3. Close and reopen VSCode
4. Attempt to run a local compilation task provided by the Particle Workbench

[image]

nzottmann, 2020-10-07:
Enabling the mentioned setting works here, but I needed one additional step:

If the PIO extension was enabled in my Particle project once, this setting does not help. The following steps worked in my existing project:
1. Check the `Activate Only On PlatformIO Project` option
2. Deactivate the PlatformIO extension
3. Restart VSCode
4. Re-enable the PlatformIO extension

I could reproduce this issue by adding a `platformio.ini` to my Particle project. As expected, this broke the Particle build, but it remained broken even after deleting the `platformio.ini` and restarting VSCode.

It seems the PlatformIO extension somehow remembers that it was active in this project once and activates itself again regardless of the option in the settings. Disabling and re-enabling the PlatformIO extension seems to clear this memory.

ivankravets, 2020-10-16:
Please update PlatformIO IDE for VSCode to 2.1.1.

Does it work now?

marc-gist, 2020-10-16:
> Please update PlatformIO IDE for VSCode to 2.1.1.
> 
> Does it work now?

No. same issue.  I disable, restart VSCode, Particle project compiles fine.  Go into PlatformIO and Enabled it, even without a restart of VSCode, compiling now fails.  Restart VSCode with PlatformIO enabled; particle compile fails with same error messages as before :(

marc-gist, 2020-10-16:
Note: if I disable PlatformIO for just the Particle Workspace/Project folder, it appears to allow me to compile fine in that workspace and open another VSCode with a PlatformIO project.
So under the Extension manager when a Workspace/Project is open, on the PlatformIO IDE settings tab, I select "Disable (Workspace)", click "Reload Required", compiling then works.

nzottmann, 2020-10-16:
2.1.1 did also not change anything for me. As I described in https://github.com/platformio/platformio-vscode-ide/issues/2045#issuecomment-705039040, if the PlatformIO extension was activated once in a Particle project, it breaks compiling until it is disabled.

To reproduce:
With the setting *Platformio-ide: Activate Only On Platform IOProject* **disabled**:
- Particle compile is always broken
- Disable PlatformIO extension
- Restart VSCode
- Particle Build works again

With the setting *Platformio-ide: Activate Only On Platform IOProject* **enabled**:
- Create a `platformio.ini` in the Particle project folder
- Restart VSCode
- Build broken
- Remove `platformio.ini`
- Restart VSCode
- Build still broken
- Disable PlatformIO extension & Restart VSCode
- Build works again
- Enable PlatformIO extension & Restart VSCode
- Build is still working

So the Particle compiler does not work if the PlatformIO extension was active anytime in the project before, which can be caused either if the setting *Platformio-ide: Activate Only On Platform IOProject* was disabled in the project or by adding a platformio.ini (which does not make sense in a Particle project, just for demonstration here).

Hope this information is helpful for your work!

ivankravets, 2020-10-16:
@busticated where is located the source code of Particle extension?

ivankravets, 2020-10-16:
How about 2.1.2 version?

marc-gist, 2020-10-16:
> How about 2.1.2 version?

Appears to be working for me! Thanks

ivankravets, 2020-10-16:
Great!

closed 2020-10-16 by ivankravets

busticated, 2020-10-16:
@ivankravets repo is private but the code ships unbundled atm so you can see if in `~/.vscode/extensions/particle.particle-vscode-core-<version-here>` :+1:

ivankravets, 2020-10-16:
@busticated it seems like the latest release has fixed this issue. Thanks for the report! ☺️

nzottmann, 2020-10-23:
Fixed it for me too! (tested with 2.1.3)

commit c019aa63 (2020-10-16): Do not patch the global environment PATH with PlatformIO // Resolve #2045, Resolve #2046