platformio/platformio-vscode-ide #2047: Start PIO Home Server

acrecido, 2020-10-02:
%23 Description of problem
 Platformio cannot load home page

  %23 Configuration

  VSCode: 1.49.2
  PIO IDE: v2.1.0
  System: Darwin, 16.7.0, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/platformio/commands/home/command.py", line 52, in cli
    inject_contrib_pysite()
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/core.py", line 77, in inject_contrib_pysite
    contrib_pysite_dir = get_core_package_dir("contrib-pysite")
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/core.py", line 38, in get_core_package_dir
    assert pm.install(spec)
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/_install.py", line 49, in install
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/_install.py", line 99, in _install
    pkg = self.install_from_registry(spec, search_filters, silent=silent)
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/_registry.py", line 83, in install_from_registry
    package = self.fetch_registry_package(spec)
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/_registry.py", line 144, in fetch_registry_package
    result = regclient.get_package(self.pkg_type, spec.owner, spec.name)
  File "/Users/ArisRecidoro/.platformio/penv/lib/python2.7/site-packages/platformio/clients/registry.py", line 145, in get_package
    if e.response.status_code == 404:
AttributeError: 'NoneType' object has no attribute 'status_code'

	at /Users/ArisRecidoro/.vscode/extensions/platformio.platformio-ide-2.1.0/node_modules/platformio-node-helpers/dist/index.js:1:12033
	at ChildProcess.s (/Users/ArisRecidoro/.vscode/extensions/platformio.platformio-ide-2.1.0/node_modules/platformio-node-helpers/dist/index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

Chompworks, 2020-10-03:
Also getting the same issue.

Trying the advice of removing the **penv** folder from .platformio hasn't resolved the issue either. 
This is on a brand new install of VS code and grabbing PlatformIO via the marketplace.

Environment:
  VSCode: 1.49.3
  PIO IDE: v2.1.0
  System: Darwin, 16.7.0, x64

Chompworks, 2020-10-03:
Output via run via terminal:

`Tool Manager: Installing platformio/contrib-pysite @ ~2.27.0
Error: Traceback (most recent call last):
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/platformio/commands/home/command.py", line 52, in cli
    inject_contrib_pysite()
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/core.py", line 77, in inject_contrib_pysite
    contrib_pysite_dir = get_core_package_dir("contrib-pysite")
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/core.py", line 38, in get_core_package_dir
    assert pm.install(spec)
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/_install.py", line 49, in install
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/_install.py", line 99, in _install
    pkg = self.install_from_registry(spec, search_filters, silent=silent)
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/_registry.py", line 83, in install_from_registry
    package = self.fetch_registry_package(spec)
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/_registry.py", line 144, in fetch_registry_package
    result = regclient.get_package(self.pkg_type, spec.owner, spec.name)
  File "/Users/drew/.platformio/penv/lib/python2.7/site-packages/platformio/clients/registry.py", line 145, in get_package
    if e.response.status_code == 404:
AttributeError: 'NoneType' object has no attribute 'status_code'
`

Chompworks, 2020-10-03:
Stumbled across this trying to dig into it more:

https://community.platformio.org/t/cant-open-pio-home/8588

It looks like the installed version of python for mac doesn't play nicely with the PlatformIO installer. The call stack above really needs to be replaced with a more user-friendly way of telling you this though!

Installing python3 via brew worked, but I also ended up installing the EOL python2 as well.

ivankravets, 2020-10-16:
Thanks! Fixed in https://github.com/platformio/platformio-core/commit/ee847e03a633a2b59cbf003a71d9fda3e1c34b52

closed 2020-10-16 by ivankravets