platformio/platformio-vscode-ide #2050: Debugger does not honor selected environment

manuelbl, 2020-10-04:
The new environment switch in the status bar is a welcome enhancement. Unfortunately, it is not honored by the debugger. Instead, the debugger uses the first environment or the one selected by `default_envs`.

As the environment switch applies to IntelliSense and the *Build* and *Upload* actions in the status bar, users expect that the debugger behaves the same. I did so too.

In my case it even lead to a very confusing situation:
- Clicking the green arrow the debugger pane started a rebuild of the project (debug configuration) – as expected.
- After that, nothing happened anymore.
- It took me quite some time until I remembered to switch to the *DEBUG CONSOLE*. I found an error basically saying *unexpected idcode: 0x0bb11477, expected 1 of 1: 0x2ba01477*.
- It took me even more time to figure out that this error message was not due to a Chinese clone of the STM32 chip, but rather due to a mismatch between the environment use by the debugger and the board connected to my computer.

I see two options to fix it:
- Honor the currently select environment when starting the debugger (preferred)
- Label the debug configuration (`launch.json`) such that it is obvious which one is used

ivankravets, 2020-10-04:
I don’t remember, do we provide different debug profiles when you switch to Debugging activity?

manuelbl, 2020-10-04:
I'm not sure I fully understand your question. But the below information might answer it. 

If you change `platformio.ini` and either rearrange the order of environments or changed `default_envs`, then `launch.json` is updated.

A configuration in `launch.json` looks like so:

```
        {
            "type": "platformio-debug",
            "request": "launch",
            "name": "PIO Debug",
            "executable": "/Users/me/Documents/my-project/.pio/build/nucleo_f042k6/firmware.elf",
            "toolchainBinDir": "/Users/me/.platformio/packages/toolchain-gccarmnoneeabi/bin",
            "svdPath": "/Users/me/.platformio/platforms/ststm32/misc/svd/STM32F042x.svd",
            "preLaunchTask": {
                "type": "PlatformIO",
                "task": "Pre-Debug"
            },
            "internalConsoleOptions": "openOnSessionStart"
        },
```
Note that the selected environment is part of the *executable* path and the board type also affects *svdPath*. So the selected environment is very relevant.