platformio/platformio-vscode-ide #2054: Start PIO Home Server After upgrade fodora to 33

dzaczek, 2020-10-07:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.3
  PIO IDE: v2.1.0
  System: Linux, 5.8.13-300.fc33.x86_64, x64

  %23 Exception
  ```
  Error: Error: Could not find the package with 'platformio/contrib-pysite @ ~2.39.0' requirements for your system 'linux_x86_64'

	at /home/dzaczek/.vscode/extensions/platformio.platformio-ide-2.1.0/node_modules/platformio-node-helpers/dist/index.js:1:12033
	at ChildProcess.s (/home/dzaczek/.vscode/extensions/platformio.platformio-ide-2.1.0/node_modules/platformio-node-helpers/dist/index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

dariomas, 2020-10-09:
Hello @dzaczek 

I have the same problem: https://github.com/platformio/platformio-vscode-ide/issues/2073

I found a **temporary** resolution:

- setting in VSCode path https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-custompath
`"platformio-ide.customPATH": "/home/xxxx/.platformio/penv/bin/"`

- disabling in VSCode builtin core https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-usebuiltinpiocore

- installing python3.8 in Fedora 33 
`dnf install python3.8`

- run the installation script with python 3.8
`python3.8 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)"`

- check into the penv/bin/ that python links to 3.8

ivankravets, 2020-10-16:
Duplicate of https://github.com/platformio/platformio-core/issues/3700

closed 2020-10-16 by ivankravets