platformio/platformio-vscode-ide #2055: Command 'PlatformIO: Home' resulted in an error (command 'platformio-ide.showHome' not found)

SanjayChakrapani, 2020-10-07:
Hello, 
I have read many issues regarding this topic, but could not find any solution
I am using PIO for 1 year, suddenly it stoped working. Im have gone through all the possible solution but not
[image]
 helping.
I have installed all the latest versions of sofwares (VSC, python, PIO)

recently i reactivated my windows 10 OS, is this cossing the issue

GunZi200, 2020-10-16:
I got this today also. I fixed it by unchecking this setting.

[image]

ivankravets, 2020-10-16:
Thanks, @GunZi200

closed 2020-10-16 by ivankravets

SanjayChakrapani, 2020-10-17:
I have that setting turned off, but not helpful

SanjayChakrapani, 2020-10-17:
today when opened, I got something like this in the output

Unable to resolve configuration with compilerPath: "C:/Users/ac-winadmin/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-gcc.exe"