platformio/platformio-vscode-ide #2067: Extension host terminated unexpectedly (VScode + PIO5)

TD-er, 2020-10-09:
### Configuration

**Operating system**:  Windows 10

**PlatformIO Version** (`platformio --version`):  5.0.1

[image]

### Description of problem

Not entirely sure if this is a problem with PIO itself or the latest VS code update.

The last few days, I experience a lot of strange issues in which the PIO actions no longer react and also navigation to function definitions does not seem to work anymore.

Not always when this happens, but quite often I do get this popup in the bottom right corner of the IDE:
[image]

For completeness, these are the installed extensions:
[image]

Not sure if it is related, but I find it hard to get adapted to the new way how PlatformIO 5 is trying to do things in VScode.
For example, as soon as you unfold an evironment in the project tasks, it tries to generate a temp .cpp file (as is normal when building an Arduino project)
[image]

However, if you click there, while it is building, the .cpp file is generated again and deleted, which causes the running build to fail.
This happens to me way more often than I would like as I also have to restart the IDE quite often the last few days (week?) which opens with all projects closed.
Could it be that this may cause stability issues with the extension host?

N.B. the logs suggest memory allocation issues, but that's probably some configured limit or some internal 32 bit limit as my system does have lots and lots of free memory right now. (currently only 13GB in use of the 96GB)

### Logs

Last logs in the "console tab" of the inspect window (after pressing "Open Developer Tools":

localProcessExtensionHost.ts:252:

```
<--- Last few GCs --->

[3360:0000031700000000]  3034793 ms: Scavenge 3797.4 (3965.5) -> 3791.5 (3966.5) MB, 4.8 / 0.0 ms  (average mu = 0.830, current mu = 0.671) allocation failure 
[3360:0000031700000000]  3034832 ms: Scavenge 3798.4 (3966.5) -> 3792.6 (3966.7) MB, 18.8 / 0.0 ms  (average mu = 0.830, current mu = 0.671) allocation failure 
[3360:0000031700000000]  3034875 ms: Scavenge 3799.0 (3966.7) -> 3793.5 (3966.7) MB, 24.0 / 0.0 ms  (average mu = 0.830, current mu = 0.671) allocation failure 

<--- JS stacktrace --->

FATAL ERROR: NewSpace::Rebalance Allocation failed - JavaScript heap out of memory
 1: 00007FF6C0686DAA VR_GetStringForHmdError+219210
 2: 00007FF6BFC143B6 node::Buffer::New+43430
 3: 00007FF6BFC14423 node::Buffer::New+43539
 4: 00007FF6BE71DFF8 v8::internal::Version::GetVersion+792
 5: 00007FF6BE71DFA1 v8::internal::Version::GetVersion+705
 6: 00007FF6BE8A9233 v8::Unlocker::~Unlocker+254403
 7: 00007FF6BE8D2351 v8::Unlocker::~Unlocker+422625
 8: 00007FF6BE8CF1F9 v8::Unlocker::~Unlocker+409993
 9: 00007FF6BE8A942C v8::Unlocker::~Unlocker+254908
10: 00007FF6BE8A6988 v8::Unlocker::~Unlocker+243992
11: 00007FF6BE8A49DD v8::Unlocker::~Unlocker+235885
12: 00007FF6BE8AFDA6 v8::Unlocker::~Unlocker+281910
13: 00007FF6BE8AFE8D v8::Unlocker::~Unlocker+282141
14: 00007FF6BE885212 v8::Unlocker::~Unlocker+106914
15: 00007FF6BEBA0AB9 v8::internal::TickSample::print+352969
16: 00007FF6BF0CAB9C v8_inspector::V8StackTraceId::ToString+2982396
17: 0000031700525C7E 
```

abstractExtensionService.ts:166

```
  ERR Extension host terminated unexpectedly. The following extensions were running: vscode.microsoft-authentication, vscode.github-authentication, ms-vscode.cpptools, vscode.debug-auto-launch, vscode.emmet, vscode.git, vscode.merge-conflict, vscode.search-result, alefragnani.Bookmarks, Atishay-Jain.All-Autocomplete, CoenraadS.bracket-pair-colorizer-2, eamodio.gitlens, LaurentTreguier.uncrustify, nicollasr.vscode-streamdeck, vscode.github, platformio.platformio-ide, vscode.configuration-editing, vscode.extension-editing, vscode.json-language-features, ms-vscode.js-debug, vscode.npm, vscode.python, ms-python.python, vscode.grunt, vscode.gulp, vscode.jake, vscode.typescript-language-features
```

webviewElement.ts:152

```
[Embedded Page] WebSocket connection to 'ws://127.0.0.1:8010/wsrpc' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/semver-umd.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/index.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/xterm.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/xterm-addon-unicode11.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/xterm-addon-search.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/index.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/windowsTerminal.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/terminal.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/eventEmitter2.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/windowsPtyAgent.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/utils.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/index.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/main.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/ExperimentationService.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/TasApiFeatureProvider.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/FilteredFeatureProvider.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/BaseFeatureProvider.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/AxiosHttpClient.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/ExperimentationServiceAutoPolling.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/ExperimentationServiceBase.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/MemoryKeyValueStorage.js.map: 
DevTools failed to load SourceMap: Could not load content for file:///C:/Program%20Files/Microsoft%20VS%20Code/resources/app/out/vs/code/electron-browser/workbench/PollingService.js.map: 
29webviewElement.ts:152 [Embedded Page] WebSocket connection to 'ws://127.0.0.1:8010/wsrpc' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
```

N.B. I first opened it [here](https://github.com/platformio/platformio-core/issues/3691), but that does not seem like the correct place.
Also at the VScode repo they referred me back to PlatformIO.

**I really want to urge to have a look at this as it is getting worse and worse by the day.**
The extension host crashes like 10+ times a day and now with the latest update of VScode it takes a few times to get PIO even running without crashing.
Loading the project tasks may already crash this, or simply loading the "home" tab or the first time I unfold an environment in my PIO config.

ivankravets, 2020-10-09:
Does it crash if you disable ALL plugins and keep only PlatformIO + C/C++?

TD-er, 2020-10-09:
Yep, just disabled all other extensions and at the first loading of VScode it crashed.

```
﻿
root-folder:1 Failed to load resource: net::ERR_FILE_NOT_FOUND
console.ts:137 [Extension Host] runCommand C:\Users\gijsn\.platformio\penv\Scripts\python.exe 
Array(2)
 
Object
localProcessExtensionHost.ts:257 Extension Host
localProcessExtensionHost.ts:258 Debugger attached.
log.ts:191  WARN UNRESPONSIVE extension host, 'ms-vscode.cpptools' took 92% of 7030ms, saved PROFILE here: 'c:\Users\gijsn\AppData\Local\Temp\exthost-f754b3.cpuprofile' 
Array(3)
localProcessExtensionHost.ts:257 Extension Host
localProcessExtensionHost.ts:258 
<--- Last few GCs --->

[8112:0000027800000000]   148647 ms: Scavenge 3923.3 (3965.5) -> 3917.4 (3965.5) MB, 27.1 / 0.0 ms  (average mu = 0.747, current mu = 0.562) allocation failure 
[8112:0000027800000000]   148703 ms: Scavenge 3923.4 (3965.5) -> 3918.2 (3965.5) MB, 38.1 / 0.0 ms  (average mu = 0.747, current mu = 0.562) allocation failure 
[8112:0000027800000000]   148772 ms: Scavenge 3923.4 (3965.5) -> 3918.9 (3965.5) MB, 52.8 / 0.0 ms  (average mu = 0.747, current mu = 0.562) allocation failure 

<--- JS stacktrace --->

FATAL ERROR: MarkCompactCollector: young object promotion failed Allocation failed - JavaScript heap out of memory
 1: 00007FF7DD256DAA VR_GetStringForHmdError+219210
 2: 00007FF7DC7E43B6 node::Buffer::New+43430
 3: 00007FF7DC7E4423 node::Buffer::New+43539
 4: 00007FF7DB2EDFF8 v8::internal::Version::GetVersion+792
 5: 00007FF7DB2EDFA1 v8::internal::Version::GetVersion+705
 6: 00007FF7DB479233 v8::Unlocker::~Unlocker+254403
 7: 00007FF7DB4C8EAC v8::Unlocker::~Unlocker+581180
 8: 00007FF7DB4B3C20 v8::Unlocker::~Unlocker+494512
 9: 00007FF7DB4B377E v8::Unlocker::~Unlocker+493326
10: 00007FF7DB4B343C v8::Unlocker::~Unlocker+492492
11: 00007FF7DB4CAEDE v8::Unlocker::~Unlocker+589422
12: 00007FF7DB48F521 v8::Unlocker::~Unlocker+345265
13: 00007FF7DC7EDAF0 node::options_parser::SplitHostPort+10656
14: 00007FF7DC7FA852 uv_thread_create_ex+354
15: 00007FF7DF25CE90 node::AsyncResource::CallbackScope::CallbackScope+21720496
16: 00007FF87BE16FD4 BaseThreadInitThunk+20
17: 00007FF87D2DCEC1 RtlUserThreadStart+33
abstractExtensionService.ts:466 Extension host terminated unexpectedly. Code:  134  Signal:  null
log.ts:197   ERR Extension host terminated unexpectedly. The following extensions were running: vscode.microsoft-authentication, ms-vscode.cpptools, vscode.typescript-language-features, vscode.debug-auto-launch, vscode.emmet, vscode.git, vscode.merge-conflict, vscode.search-result, LaurentTreguier.uncrustify, vscode.github, platformio.platformio-ide, ms-vscode.js-debug, vscode.configuration-editing, vscode.extension-editing, vscode.json-language-features, vscode.npm, vscode.github-authentication
notificationsAlerts.ts:40 Extension host terminated unexpectedly.
webviewElement.ts:150 [Embedded Page] WebSocket connection to 'ws://127.0.0.1:8010/wsrpc' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
```

TD-er, 2020-10-09:
Just use this repo to make it crash: https://github.com/letscontrolit/ESPEasy

ivankravets, 2020-10-09:
This is the first report since August 2020. We will investigate soon. Thanks for the report. 

P.S: I highly recommend moving away from the INO extension and use native CPP. At a minimum, I don't see a reason to use PlatformIO + VSCode when files are organized as INO. C++ extension does not parse them correct - they are fake C++ files.

TD-er, 2020-10-09:
I'm working very very hard on that transition, but it takes a while to finish.

I do know (by now, found out the hard way) what other issues working with .ino cause, so believe me I am moving away from .ino as fast as I can.

ivankravets, 2020-10-09:
Yes, I understand. In any case, we will take a look at it. The interesting thing that we don't use any computations with Node.Js that can lead to 4Gb memory limit, we just call PlatormIO Core CLI as subprocess.

ivankravets, 2020-10-09:
I forgot. Could you disable C/C++ extension? Just leave ONLY PlatformIO. Maybe, this extension crashes with INO files?

TD-er, 2020-10-09:
Ah sorry, thought it was disabled... Will check now.

TD-er, 2020-10-09:
[image]

Cannot disable it...

ivankravets, 2020-10-09:
just uninstall it

TD-er, 2020-10-09:
[image]

So I guess I have to disable PIO first to do so?

ivankravets, 2020-10-09:
You can do the opposite. Remove PlatformIO extension and leave only C++. Restart VSCode.

TD-er, 2020-10-09:
I have C/C++ now disabled and PIO is now really the only one active and it does seem a bit more stable (not yet got extension host to report it has crashed)
But loading the libraries update tab gives this error:
```
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n File \"C:\\Users\\gijsn\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n cli() # pylint: disable=no-value-for-parameter\r\n File \"C:\\Users\\gijsn\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n return self.main(*args, **kwargs)\r\n File \"C:\\Users\\gijsn\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n rv = self.invoke(ctx)\r\n File \"C:\\Users\\gijsn\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n return super(PlatformioCLI, self).invoke(ctx)\r\n File \"C:\\Users\\gijsn\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n File \"C:\\Users\\gijsn\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n File \"C:\\Users\\gijsn\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n return ctx.invoke(self.callback, **ctx.params)\r\n File \"C:\\Users\\gijsn\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n return callback(*args, **kwargs)\r\n File \"C:\\Users\\gijsn\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\r\n return f(get_current_context(), *args, **kwargs)\r\n File \"C:\\Users\\gijsn\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib\\command.py\", line 247, in lib_update\r\n lm = LibraryPackageManager(storage_dir)\r\n File \"C:\\Users\\gijsn\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\library.py\", line 30, in __init__\r\n PackageType.LIBRARY, package_dir or get_project_global_lib_dir()\r\n File \"C:\\Users\\gijsn\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\base.py\", line 54, in __init__\r\n self.package_dir = self.ensure_dir_exists(package_dir)\r\n File \"C:\\Users\\gijsn\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\base.py\", line 98, in ensure_dir_exists\r\n assert os.path.isdir(path)\r\nAssertionError\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"
```

[image]

Could this be related or an indication something is wrong with my PIO config/installation?

ivankravets, 2020-10-09:
Try the latest PIO Core Dev (`pio upgrade --dev`) and PlatformIO should be stable.

So, as I expected, the issue is caused by the C++ extension. They need a ton of resources to parse your sources to build meta DB.

You can try now the opposite, to remove the PlatformIO extension and use only C++, open the same project. Then, search in command "Rebuild C++ DB". It should crash.

TD-er, 2020-10-09:
I am now running with PIO uninstalled and C/C++ active and so far I cannot get it to crash.

TD-er, 2020-10-09:
> You can try now the opposite, to remove the PlatformIO extension and use only C++, open the same project. Then, search in command "Rebuild C++ DB". It should crash.

Looking on how to rebuild that... not found it so far.

ivankravets, 2020-10-09:
[image]

TD-er, 2020-10-09:
[image]

You mean "rescan workspace" ?

ivankravets, 2020-10-09:
Reset IntelliSense DB

TD-er, 2020-10-09:
Doesn't seem to crash.
Could it be the combination of opening the PIO task list which does generate a temp .cpp file, which causes havoc on the intellisense?

TD-er, 2020-10-09:
Installed PIO and updated to the last dev.
[image]

Just unfolding the 3rd project task was already crashing the extension host:
[image]

N.B. I did wait a while before opening the next one as it will otherwise cause issues as the temp .cpp is deleted by the previous one before the next one has finished.

closed 2021-01-07 by stale[bot]