platformio/platformio-vscode-ide #2090: Start PIO Home Server

gritJack, 2020-10-11:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.0
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Error: Could not find the package with 'platformio/contrib-pysite @ ~2.39.0' requirements for your system 'windows_amd64'

	at c:\Users\LiuLingJian\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\index.js:1:12033
	at ChildProcess.s (c:\Users\LiuLingJian\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\LiuLingJian\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-10-16:
Duplicate of https://github.com/platformio/platformio-core/issues/3700

closed 2020-10-16 by ivankravets

itschina, 2022-07-20:
Error: Could not start PIO Home server: Error: timeout at c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:199:72...