platformio/platformio-vscode-ide #2120: Could not find the package with 'platformio/contrib-pysite @ ~2.39.0' requirements for your system 'windows_amd64'

gaafouri, 2020-10-13:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.0
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.15063, x64

  %23 Exception
  ```
  Error: Error: Could not find the package with 'platformio/contrib-pysite @ ~2.39.0' requirements for your system 'windows_amd64'

	at c:\Users\Ahmed\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\index.js:1:12033
	at ChildProcess.s (c:\Users\Ahmed\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\Ahmed\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

gaafouri, 2020-10-13:
PS C:\Users\Ahmed\Documents\PlatformIO\Projects\arduino nano> pio home
********************************************************************************************************************************************************************************************************
Obsolete PIO Core v5.0.1 is used (previous was 5.0.2b2)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-platformio-cores-in-a-system
********************************************************************************************************************************************************************************************************
Tool Manager: Installing platformio/contrib-pysite @ ~2.39.0
Error: Could not find the package with 'platformio/contrib-pysite @ ~2.39.0' requirements for your system 'windows_amd64'
PS C:\Users\Ahmed\Documents\PlatformIO\Projects\arduino nano>

ivankravets, 2020-10-16:
Duplicate of https://github.com/platformio/platformio-core/issues/3700

closed 2020-10-16 by ivankravets