platformio/platformio-vscode-ide #2126: Start PIO Home Server

CaioMonteiroICTS, 2020-10-13:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.0
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Webview is disposed
	at b.assertNotDisposed (c:\Users\CTS\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:836)
	at b.set html [as html] (c:\Users\CTS\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:225)
	at u.newPanel (c:\Users\CTS\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\extension.js:1:26272)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async u.toggle (c:\Users\CTS\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\extension.js:1:25794)
  ```

ivankravets, 2020-10-15:
How to reproduce this issue?

CaioMonteiroICTS, 2020-10-15:
I don't know how to reproduce the problem. It happened a few times just to
start VSCode. I noticed an update and the error only happened once during
this period. It's random.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Livre
de vírus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

ivankravets, 2020-10-15:
Do you mean that you can not reproduce it again?

CaioMonteiroICTS, 2020-10-15:
It's random. When it happened, I just restarted the VSC. But after I
updated things, the error stopped happening. If it happens again I will
report.

ivankravets, 2020-10-15:
@mjbvz, sorry that disturb you. Do you have any ideas? The lastest VSCode 1.50 totally flooded our repository:  https://github.com/platformio/platformio-vscode-ide/issues

How can we fix it? I can't reproduce it on my macOS.

mjbvz, 2020-10-15:
This looks like a result of a bug fix in VS Code that I made so that webviews  properly track when they are disposed of: https://github.com/microsoft/vscode/commit/d29fc5038adfa343059c99cddaec13dd1da1ef33 

Can you make sure the webviews you create are tracking disposal properly. One possible cause I see is this line: https://github.com/platformio/platformio-vscode-ide/blob/19ea0a3fc5fb04c466729726dddb2fe5488928ba/src/home.js#L68

Since getting the html is async, `panel` could be disposed by the time you try setting `.html` on it.

Take a look at your code first since we have not yet seen other reports of this issue. If this looks like a VS Code bug, could you put together a small example that demonstrates the issue?

CaioMonteiroICTS, 2020-10-15:
I wish I could help them, but I don't know how. If the error happens again I will be more attentive to report better. But I appreciate the readiness. Thank you!

CaioMonteiroICTS, 2020-10-15:
So ... The error happened again. I can't say if it's the same. But it just happens to open the VCS. I used VSC to open an empty html that I was going to start editing now and the error came up. Eu preferi reportar aqui para não abrir outra solicitação.

%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Webview is disposed
	at b.assertNotDisposed (c:\Users\CTS\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:836)
	at b.set html [as html] (c:\Users\CTS\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:225)
	at u.newPanel (c:\Users\CTS\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\extension.js:1:26272)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async u.toggle (c:\Users\CTS\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\extension.js:1:25794)
  ```

ivankravets, 2020-10-15:
Do you use antivirus tool?

CaioMonteiroICTS, 2020-10-15:
I have an antivirus. Do you believe that it may be generating the conflict?

ivankravets, 2020-10-15:
This our issue, not VSCode. We should inform user that someone killed/blocked PIO Home server. This is the reason why you don’t see PlatformIO Home in VSCode. 

We will work on a workaround for this issue.

closed 2020-10-16 by ivankravets

ivankravets, 2020-10-16:
Please update PlatformIO IDE for VSCode to 2.1.1

commit 3ccda29e (2020-10-16): Fixed an issue with "Webview is disposed" // Resolve #2126