platformio/platformio-vscode-ide #2135: Start PIO Home Server

mxmka1000, 2020-10-14:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.0
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.20231, x64

  %23 Exception
  ```
  Error: Webview is disposed
    at b.assertNotDisposed (c:\Users\maxim\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:836)
    at b.set html [as html] (c:\Users\maxim\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:225)
    at u.newPanel (c:\Users\maxim\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\webpack:\src\home.js:68:21)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:94:5)
    at u.toggle (c:\Users\maxim\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\webpack:\src\home.js:45:10)
  ```

ivankravets, 2020-10-14:
How to reproduce it?

ivankravets, 2020-10-15:
How to reproduce this issue?

closed 2020-10-16 by ivankravets