platformio/platformio-vscode-ide #2136: Start PIO Home Server

David-Nam, 2020-10-14:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.0
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Webview is disposed
	at b.assertNotDisposed (c:\Users\deity\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:836)
	at b.set html [as html] (c:\Users\deity\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:225)
	at u.newPanel (c:\Users\deity\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\extension.js:1:26272)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async u.toggle (c:\Users\deity\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\extension.js:1:25794)
  ```

ivankravets, 2020-10-15:
v

ivankravets, 2020-10-15:
How to reproduce this issue?

closed 2020-10-16 by ivankravets