platformio/platformio-vscode-ide #2143: Start PIO Home Server

robocog, 2020-10-14:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.0
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Webview is disposed
	at b.assertNotDisposed (c:\Users\admin\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:836)
	at b.set html [as html] (c:\Users\admin\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:225)
	at u.newPanel (c:\Users\admin\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\extension.js:1:26272)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async u.toggle (c:\Users\admin\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\extension.js:1:25794)
  ```

ivankravets, 2020-10-15:
How to reproduce this issue?

robocog, 2020-10-15:
Problem seems to have gone now :D
It was just coming up with that error every time I opened VScode to edit Marlin firmware

I have 3 workspaces - one for each printer - all 3 were doing it - but sill compiled the firmware regardless of the message so it wasn't stopping the software from working

I have tried all 3 just now and the message no longer pops up and I haven't changed anything my end

Thanks
Regards
Rob

closed 2020-10-15 by robocog