platformio/platformio-vscode-ide #2144: Start PIO Home Server

jgomezcic, 2020-10-15:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Webview is disposed
	at b.assertNotDisposed (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:836)
	at b.set html [as html] (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:225)
	at u.newPanel (c:\Users\JULIAN\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\extension.js:1:26272)
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async u.toggle (c:\Users\JULIAN\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\extension.js:1:25794)
  ```

ivankravets, 2020-10-15:
How to reproduce this issue?

jgomezcic, 2020-10-15:
> How to reproduce this issue?

Hi, it occurred in the VSCode upgrade

ivankravets, 2020-10-15:
Could you reproduce it again?

closed 2020-10-16 by ivankravets