platformio/platformio-vscode-ide #2146: Start PIO Home Server

jones3511, 2020-10-15:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Webview is disposed
	at b.assertNotDisposed (c:\Users\Brian\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:836)
	at b.set html [as html] (c:\Users\Brian\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:225)
	at u.newPanel (c:\Users\Brian\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\extension.js:1:26272)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async u.toggle (c:\Users\Brian\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\extension.js:1:25794)
  ```

ivankravets, 2020-10-15:
How to reproduce this issue?

jones3511, 2020-10-15:
Something is wrong with the bugfix 2.0 branch of the configurations for the
ender 5 pro 32 bit board. Not sure what but I was able to build the
firmware using another config from the github repository.

ivankravets, 2020-10-15:
The issue is related to VSCode. Could you reproduce it again?

closed 2020-10-16 by ivankravets