platformio/platformio-vscode-ide #2163: Start PIO Home Server

David-Nam, 2020-10-16:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Error: Could not find the package with 'platformio/contrib-pysite @ ~2.39.0' requirements for your system 'windows_amd64'

    at args (c:\Users\deity\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:196:9)
    at ChildProcess.s (c:\Users\deity\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:132:7)
    at ChildProcess.emit (events.js:223:5)
    at ChildProcess.cp.emit (c:\Users\deity\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:1021:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

Dev-I-J, 2020-10-16:
Facing the same error.

ivankravets, 2020-10-16:
Duplicate of https://github.com/platformio/platformio-core/issues/3700

closed 2020-10-16 by ivankravets