platformio/platformio-vscode-ide #2170: PIO 2.1.1 does not work

BbIKTOP, 2020-10-16:
VSCode
Version: 1.50.1
Commit: d2e414d9e4239a252d1ab117bd7067f125afd80a
Date: 2020-10-13T14:53:05.704Z (3 days ago)
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Darwin x64 19.6.0

Installed PIO stopped working after upgrade.
While attempting to reinstall, PIO does not downloads all required files and still not available. Downgrading to 2.1.0 solves this problem immediately.

jmpmscorp, 2020-10-16:
Same happens here. Downgraded to 2.1.0 and everything works again.

Kortivex, 2020-10-16:
Same here. Reverted to v2.1.0 and everything works again.

> VS Code Version 1.50.1 on Windows 10 2004

SwedishChef, 2020-10-16:
> Same here. Reverted to v2.1.0 and everything works again.

Could you please tell how to downgrade? THX

ijulanday, 2020-10-16:
Also having issues with PIO 2.1.1, I'm getting this error in the the Dev Tools console from abstractExtensionService.ts:

`Activating extension 'platformio.platformio-ide' failed: Cannot find module 'os-tmpdir'
Require stack:
- /Users/admin/.vscode/extensions/platformio.platformio-ide-2.1.1/node_modules/tmp/lib/tmp.js
- /Users/admin/.vscode/extensions/platformio.platformio-ide-2.1.1/node_modules/platformio-node-helpers/dist/index.js
- /Users/admin/.vscode/extensions/platformio.platformio-ide-2.1.1/dist/extension.js
- /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js
- /Applications/Visual Studio Code.app/Contents/Resources/app/out/bootstrap-amd.js
- /Applications/Visual Studio Code.app/Contents/Resources/app/out/bootstrap-fork.`

@SwedishChef go to the Extensions window and click the gear on the PIO extension, then click Install Another Version

SwedishChef, 2020-10-16:
@ijulanday - so easy ;-) thanks a lot!

thegoodhen, 2020-10-16:
Also can confirm, same error message.

thegoodhen, 2020-10-16:
> > Same here. Reverted to v2.1.0 and everything works again.
> 
> Could you please tell how to downgrade? THX

Click on the "extensions" icon on the left vertical side menu of VSCode. On the Platformio item, click on the small gear icon and then on "Install another version."

niharG8, 2020-10-16:
Same issue here on macOS and Windows. PIO 2.1.1 is DOA with VS Code 1.50.1

ivankravets, 2020-10-16:
Sorry for the issue. Soem idiot removed dependency to `os-tmpdir` and didn't increment the major version. As result, it affected other packages. Indeed, we don't use this package. 

So, just added "permanently" and released PlatformIO IDE 2.1.2. Please update.

P.S: Similar case https://qz.com/646467/how-one-programmer-broke-the-internet-by-deleting-a-tiny-piece-of-code/

closed 2020-10-16 by ivankravets