platformio/platformio-vscode-ide #2171: Lock upload when RAM usage is exceeded

n4nn31355, 2020-10-16:
I think it would be nice feature because it wasn't a very fun time trying to figure out why my board turns to a brick after upload.

ivankravets, 2020-10-17:
It's already implemented. What is your dev-board?

n4nn31355, 2020-10-17:
My board is sparkfun promicro 5V clone from aliexpress.

I found that the build fails (and therefore the upload does not continue) when the Flash size is exceeded, but when the RAM is exceeded, we get a successful build and following upload:
[image]

But I also found that my board is not working even if the RAM usage shows `98%`. So now it looks to me like it's a board issue or expected behavior. I struggled to find a way to set the RAM threshold to fail build, but couldn't find it.
I still think that it would be a nice feature or if it already exists, it should be documented.

closed 2020-12-23 by stale[bot]