platformio/platformio-vscode-ide #2179: platformio : The term 'platformio' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a pa  th was included, verify that the path is correct and try again.

infatum1, 2020-10-18:
Creality Ender 5 (screenshot - https://photos.app.goo.gl/BXVKtJ8CBb7fG5989) or in text form:

platformio : The term 'platformio' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is correct and try again.

At line:1 char:1
+ platformio run -e melzi
+ ~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (platformio:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
 
PS C:\Users\   \Downloads\Marlin-2.0.x (1)\Marlin-2.0.x> echo "done" >C:\Users\    \AppData\Local\Temp\ipc
PS C:\Users\    \Downloads\Marlin-2.0.x (1)\Marlin-2.0.x>

Tried uninstalling and reinstalling. Same persists. Oddly enough, ivankravets states he fixed this bug (https://community.platformio.org/t/platformio-is-not-recognized-as-an-internal-or-external-command/431/22)
I guess not.

DrDrache, 2020-10-18:
I am having the same exact issue - old projects run fine - anything new does not.
Vscode insiders.
have removed and reinstalled the platformIO IDE extension.

Migamix, 2020-10-19:
revert and stick to version 2.1.0 until people confirm the issue is "FULLY" fixed
this caused me a to waste time debugging the compiler and not my firmware.

sebby49, 2020-10-20:
i reverted back to 2.1.0 and still having the same issue ugh

closed 2020-12-26 by stale[bot]