platformio/platformio-vscode-ide #2181: Start PIO Home Server

emibia, 2020-10-18:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.1.3
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Error: Could not find the package with 'platformio/contrib-pysite @ ~2.39.0' requirements for your system 'windows_amd64'

	at c:\Users\emidi\.vscode\extensions\platformio.platformio-ide-2.1.3\node_modules\platformio-node-helpers\dist\index.js:1:5625
	at ChildProcess.c (c:\Users\emidi\.vscode\extensions\platformio.platformio-ide-2.1.3\node_modules\platformio-node-helpers\dist\index.js:1:878983)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\emidi\.vscode\extensions\platformio.platformio-ide-2.1.3\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

hb020, 2020-12-24:
Potential duplicate of #148

ivankravets, 2021-01-20:
https://github.com/platformio/platformio-vscode-ide/issues/148

closed 2021-01-20 by ivankravets