platformio/platformio-vscode-ide #2182: The terminal process failed to launch: Path to shell executable "platformio" is not a file of a symlink.

IanMercer, 2020-10-19:
I recently upgraded Ubuntu to 20.04 and now I cannot get platformio to work in VSCode.  Hitting the Platformio:Build button results in an error `The terminal process failed to launch: Path to shell executable "platformio" is not a file of a symlink.`

Running `platformio` or `pio` in the same Terminal Window runs just fine.

I have uninstalled and reinstalled platformio.
I have tried deleting ~/.platformio and reinstalling.
I have tried both with and without `platformio-ide.useBuiltinPIOCore`.
I have tried setting `platformio-ide.customPATH` to match my Linux $PATH.

VSCode: 1.50.1
Platformio: Core 5.0.1 Home 3.3.0

IanMercer, 2020-12-19:
What details do you need?

ivankravets, 2021-01-05:
1. Please remove `~/.platformio/penv` folder
2. Enable back  `platformio-ide.useBuiltinPIOCore` and make empty `platformio-ide.customPATH`
3. Check that you have Python 3.6+ in your system via `python --version`
4. Restart VSCode.

closed 2021-01-05 by ivankravets