platformio/platformio-vscode-ide #2190: PIO Home not working on port 8008

emphyrion, 2020-10-22:
Hello,
I'm using Platformio in VSCode on Windows 10.
It used to work until two days ago. It is now impossible to get the pio home page.
If I type "pio home" in the terminal, I get an error in my brower and the terminal says:
Couldn't listen on 127.0.0.1:8008: [WinError 10013]

If I type "pio home --port 8888", it works!
But now? What should I do to fix that problem?

Thank you for your help.

emphyrion, 2020-10-22:
OK, I uninstalled completly VSCode and PlateformIO and reinstalled everything. And it fixed the problem.
I still would have liked to know what happened and don't want to reinstalled again next time it happend...

closed 2020-12-29 by stale[bot]

DanMan32, 2023-01-22:
I also came across the problem where in VS Code, platformIO home doesn't work, nothing listening on 8008.
I'm seeing if uninstalling/reinstalling PlatformIO extension in VS fixes the problem.

DanMan32, 2023-01-22:
Well, that didn't work, not even after disabling Windows Firewall.