platformio/platformio-vscode-ide #2192: Start PIO Home Server

drmulligan, 2020-10-23:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.0
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Webview is disposed
	at b.assertNotDisposed (c:\Users\David\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:836)
	at b.set html [as html] (c:\Users\David\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:840:225)
	at u.newPanel (c:\Users\David\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\extension.js:1:26272)
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async u.toggle (c:\Users\David\.vscode\extensions\platformio.platformio-ide-2.1.0\dist\extension.js:1:25794)
  ```

hb020, 2020-12-24:
Potential duplicate of #148

ivankravets, 2021-01-05:
Please upgrade the PlatformIO IDE extension in VSCode to the latest version. The issue is already fixed.

closed 2021-01-05 by ivankravets