platformio/platformio-vscode-ide #2193: Start PIO Home Server

nebinscratch, 2020-10-23:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.1.3
  System: Darwin, 19.6.0, x64

  %23 Exception
  ```
  Error: Error: You are not connected to the Internet.
PlatformIO needs the Internet connection to download dependent packages or to work with PlatformIO Account.

	at /Users/devinnamaky/.vscode/extensions/platformio.platformio-ide-2.1.3/node_modules/platformio-node-helpers/dist/index.js:1:5625
	at ChildProcess.c (/Users/devinnamaky/.vscode/extensions/platformio.platformio-ide-2.1.3/node_modules/platformio-node-helpers/dist/index.js:1:878983)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

nebinscratch, 2020-10-23:
Cannot get any PIO screens to load. Already restarted VSCode and also restarted the entire computer. Simply won't connect even though I am connected to the internet.

DrNebin, 2020-10-29:
Installed Atom and Platform IDE which worked, then reinstalled VSCODE and Platform for VSCODE and is working now.

closed 2020-10-29 by DrNebin