platformio/platformio-vscode-ide #2202: Start PIO Home Server

bernateau, 2020-10-28:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.1.3
  System: Darwin, 16.7.0, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/platformio/commands/home/command.py", line 52, in cli
    inject_contrib_pysite()
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/core.py", line 77, in inject_contrib_pysite
    contrib_pysite_dir = get_core_package_dir("contrib-pysite")
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/core.py", line 38, in get_core_package_dir
    assert pm.install(spec)
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/_install.py", line 49, in install
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/_install.py", line 99, in _install
    pkg = self.install_from_registry(spec, search_filters, silent=silent)
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/_registry.py", line 83, in install_from_registry
    package = self.fetch_registry_package(spec)
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/platformio/package/manager/_registry.py", line 144, in fetch_registry_package
    result = regclient.get_package(self.pkg_type, spec.owner, spec.name)
  File "/Users/fredericbernateau/.platformio/penv/lib/python2.7/site-packages/platformio/clients/registry.py", line 145, in get_package
    if e.response.status_code == 404:
AttributeError: 'NoneType' object has no attribute 'status_code'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at /Users/fredericbernateau/.vscode/extensions/platformio.platformio-ide-2.1.3/node_modules/platformio-node-helpers/dist/index.js:1:5625
	at ChildProcess.c (/Users/fredericbernateau/.vscode/extensions/platformio.platformio-ide-2.1.3/node_modules/platformio-node-helpers/dist/index.js:1:878983)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Socket.<anonymous> (internal/child_process.js:430:11)
	at Socket.emit (events.js:223:5)
	at Pipe.<anonymous> (net.js:664:12)
  ```

hb020, 2020-12-24:
Potential duplicate of #148

ivankravets, 2021-01-20:
Please remove `/Users/fredericbernateau/.platformio/penv` folder and restart VSCode.

closed 2021-01-20 by ivankravets