platformio/platformio-vscode-ide #2203: Debugger does not honor selected environment

manuelbl, 2020-10-28:
The new environment switch in the status bar is a welcome enhancement. Unfortunately, it is not honored by the debugger. Instead, the debugger uses the first environment or the one selected by `default_envs`.

As the environment switch applies to IntelliSense and the *Build* and *Upload* actions in the status bar, users expect that the debugger behaves the same. I did so too.

In my case it even lead to a very confusing situation:
- Clicking the green arrow the debugger pane started a rebuild of the project (debug configuration) – as expected.
- After that, nothing happened anymore.
- It took me quite some time until I remembered to switch to the *DEBUG CONSOLE*. I found an error basically saying *unexpected idcode: 0x0bb11477, expected 1 of 1: 0x2ba01477*.
- It took me even more time to figure out that this error message was not due to a Chinese clone of the STM32 chip, but rather due to a mismatch between the environment use by the debugger and the board connected to my computer.

I see two options to fix it:
- Honor the currently select environment when starting the debugger (preferred)
- Label the debug configuration (`launch.json`) such that it is obvious which one is used

ivankravets, 2020-10-31:
https://github.com/platformio/platformio-vscode-ide/issues/2018#issuecomment-719995422

You will also need the latest PlatformIO Core 5.0.3 Dev.

Does it work now? Thanks!

P.S: Also would be thankful for the feedback with upcoming chnages in 2.2.0.

manuelbl, 2020-11-01:
I've jus tried it and it didn't work. The correct environment was used to build a debug version. But the debugger is then started with a different (incorrect) environment.

`launch.json` looks ok. But the first line in the *DEBUG CONSOLE* already points at the incorrect environment (the selected environment is *nucleo-f303re*, the debugger however used *nucleo_f042k6*, which is the top most in `platformio.ini`):

```
Processing nucleo-f042k6 (board: nucleo_f042k6; platform: ststm32; framework: libopencm3)
```

Strangely enough, it looks as if the target was build twice: the output of the correct one appears in the *TERMINAL* view, the output of the incorrect one appears in *DEBUG CONSOLE*.

ivankravets, 2020-11-01:
Do you use the latest PlatformIO Core 5.0.3 Dev?

manuelbl, 2020-11-01:
Yes, I do.

And I'm having more troubles. On a project with a single environment, I can't start the debugger anymore. I always get the error message "Could not find the specified task." This is new. It has worked before I upgraded to the development version.

And I don't know if it is related... Quite often *Build* and *Upload* in the status bar don't work. Instead of executing something, they just display a pop up (the kind VSC displays to let you select something): 

[image]

This isn't new. It has occurred from time to time since the environment selector in the status bar was introduced. The workaround is to again select the already selected environment from the status bar.

ivankravets, 2020-11-01:
Keep PlatformIO IDe extension at 2.2.0-beta and downgrade PlatformIO Core to the latest stable 5.0.2. I'll try to investigate this issue.

Another workaround is to use `"PIO Debug (skip Pre-Debug)"` instead of `"PIO Debug"`.

manuelbl, 2020-11-01:
I have been using the new core 5.0.3a1 but the old extension (2.1.3). How do I upgrade to the beta version?

ivankravets, 2020-11-01:
See instruction by a link above. Thanks.

ivankravets, 2020-11-03:
2.2.0 has been released! You will just need the latest PIO Core 5.0.3-dev.

manuelbl, 2020-11-03:
It looks good. For the first time, debugging the selected environment works. Thanks.

The "No configured task..." pop-up didn't appear anymore. I hope it has been fixed for good.

I can still generate the "Could not find the specified task." error. It seems to be related to currently open file in the editor. The error occurs if a file from another project is open. The solution is to open a file from the project to be debugged (or bring the file to the front).

ivankravets, 2020-11-04:
VSCode consumes tasks depending on the active project. We don’t rebuild index every time. It seems that we should provide debugging configuration using VSCode API and not use launch.json. 

Other solution is to revert back PlatformIO Core and use a base pre-debug task.

maxgerhardt, 2021-02-02:
Also noted as a bug in [community thread](https://community.platformio.org/t/pio-debug-doesnt-work-when-specifying-default-envs-in-ini-file/19053?u=maxgerhardt)

GadgetAngel, 2021-02-06:
THANK YOU

I moved to "platformIO dev" branch and I also checked that I had the latest "PlafformIO IDE" extension installed.  Moving to the dev branch,  solved my **"Error could not find specified task"**

Thankyou!

commit ce045446 (2020-10-31): Fixed an issue when debugger does not honor selected environment // Resolve #2203