platformio/platformio-vscode-ide #2206: Change Upload USB Type and Port

ijulanday, 2020-10-30:
Hi,

I'm writing a program fora Teensy 4.0 board using ChibiOS (ChRt), but I can't seem to upload the code successfully with Platformio (Core 5.0.1 Home 3.3.1 on macOS Catalina 10.15.7).

I was able to run build and upload my code with the Teensyduino plugin for the Arduino IDE, but I had to make some adjustments: under "Tools > USB Type", I changed the selection from "Serial" to "Raw HID". After which, I was able to successfully upload and bring up a serial monitor. However, the port that is receiving Serial output from the Teensy was titled "HID=16c0:0478.ffc9.4", and as a "Teensy Port" under "Tools > Port". More specifically, if I run `system_profiler SPUSBDataType` I get the following entry:

```
Teensyduino RawHID:

          Product ID: 0x0486
          Vendor ID: 0x16c0
          Version: 2.79
          Serial Number: 7498000
          Speed: Up to 480 Mb/s
          Manufacturer: Teensyduino
          Location ID: 0x04100000 / 8
          Current Available (mA): 500
          Current Required (mA): 100
          Extra Operating Current (mA): 0
```

My guess is that this has to do with the Teensy needing HID protocol for uploading, as listed under the PJRC troubleshooting page (https://www.pjrc.com/teensy/troubleshoot.html). **I would like to know do this, or something similar, in PlatformIO.** I apologize if I'm a bit rambling, I'm not very well informed with what's going on behind the scenes when I press the "upload" button.

Below is the program I'm running and some pictures.

```c
#include <Arduino.h>
#include "ChRt.h"
 
// working area
static THD_WORKING_AREA(waThreadHEART, 64);
// actual function
static THD_FUNCTION(ThreadHEART, arg)
{
  (void)arg;
  //basic heartbeat thread to tell us whether weve crashed
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
  while (1)
  {
    digitalWrite(LED_BUILTIN, HIGH);
    chThdSleepMilliseconds(83);
    digitalWrite(LED_BUILTIN, LOW);
    chThdSleepMilliseconds(166);
    digitalWrite(LED_BUILTIN, HIGH);
    chThdSleepMilliseconds(125);
    digitalWrite(LED_BUILTIN, LOW);
    Serial.print(".");
    chThdSleepMilliseconds(626);
    
  }
}
 
void chStartup()
{
  chThdCreateStatic(waThreadHEART, sizeof(waThreadHEART),
                    NORMALPRIO + 1, ThreadHEART, NULL); 
}
 
void setup()
{
  Serial.begin(9600);
  while (!Serial)
  {
    delay(1);
  }
  
  Serial.println("Serial was started!");

  chBegin(chStartup);
  while(1) {}
}
 
void loop()
{
}
```

[image]

ijulanday, 2020-10-30:
Welp, I might have posted this a bit hastily. I removed the library dependency and instead just cloned the source and stuck it under the /lib directory, as well as specifying `upload_protocol = teensy-gui` in my `platformio.ini` file (although that's what's run normally, so I don't think that did anything). So now I guess the issue lies with installing ChRt from the PIO Home Library Registry page.

ivankravets, 2021-01-05:
See https://docs.platformio.org/en/latest/platforms/teensy.html

closed 2021-01-05 by ivankravets