platformio/platformio-vscode-ide #2207: Start PIO Home Server

Okkam007, 2020-10-31:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.1.3
  System: Linux, 4.19.0-12-amd64, x64

  %23 Exception
  ```
  Error: 
Aborted!

	at /home/arduino1/.vscode/extensions/platformio.platformio-ide-2.1.3/node_modules/platformio-node-helpers/dist/index.js:1:5625
	at ChildProcess.c (/home/arduino1/.vscode/extensions/platformio.platformio-ide-2.1.3/node_modules/platformio-node-helpers/dist/index.js:1:878983)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Socket.<anonymous> (internal/child_process.js:430:11)
	at Socket.emit (events.js:223:5)
	at Pipe.<anonymous> (net.js:664:12)
  ```

Okkam007, 2020-10-31:
please help me launch

hb020, 2020-12-24:
Potential duplicate of #148

ivankravets, 2021-01-20:
Have you resolved this issue?

closed 2021-02-02 by ivankravets